ARMmbed/mbed-os #5573: Windows USB resets board at 15 minute intervals!?

RobMeades, 2017-11-23:
I am developing an application on an Mbed C030 board which happens to log diagnostics behaviour to file and send it off to a server when the board comes up after a reset. When I looked at my data this morning I found that the application had reset every 15 minutes, the STM32F4 MCU reporting that it was reset by an external pin.  More than that, it had reset at 00, 15, 30 and 45 minutes past the hour, and my application is **not** time synchronised.

If I unplug the USB cable that is connected to my Windows machine from the debug interface, and instead supply 5V directly to the board, the application no longer resets and all is fine.

Now I've just moved to a new laptop/Win10, so I can't be sure if this has always been happening and I've just not had an application that would pick it up.  I have all the "don't power me down" boxes ticked on all the USB interfaces and Windows Event logs don't show anything of interest happening.

Has anyone else come across such behaviour, or am I the unlucky one?  If the behaviour is known, any ideas on how I stop it!?

RobMeades, 2017-11-30:
I'll take that as a "no" then :-).

cmonr, 2017-12-06:
Hi @RobMeades,

Were you able to get to the bottom of the reset issue? Have you tried powering the board from Windows without the debugging interface in use?

RobMeades, 2017-12-06:
Not yet got to the bottom of it. I ended up attaching a USB sniffer to the interface and that clearly shows a USB reset control message being sent by the Win10 PC every 15 minutes. The board uses an ST debug chip and so I've sent the USB sniffer log to ST so that they can investigate what's going on.

cmonr, 2017-12-06:
Sounds good. Since it sounds like the issue isn't related to mbed-os, I'm going to close the issue. 

Feel free to create another issue if it turns out to not be the case.

closed 2017-12-06 by cmonr

RobMeades, 2017-12-15:
FYI, it _looks_ as though this may be caused by the Dell Support Assist Agent interacting badly with the STLink USB driver for the board.  Having uninstalled the Dell Support Assist Agent I believe the problem has gone away but I'll leave it a little longer to be completely sure.