ARMmbed/mbed-os #5575: travis: The job exceeded the maximum time limit for jobs

0xc0170, 2017-11-24:
I've have noticed few failures due to maximum time limit for jobs in travis. The second run is always green though, usually restarting a job makes it overcome this limit. The last build failure due to timeout was here: https://github.com/ARMmbed/mbed-os/pull/5273#issuecomment-346754532. The message in the log is:

> The job exceeded the maximum time limit for jobs, and has been terminated.

What I found in the travis doc, this should help: https://docs.travis-ci.com/user/common-build-problems/#My-builds-are-timing-out

cc @theotherjimmy @adbridge @bridadan

adbridge, 2017-11-24:
Looks like there are a few options there we can try!

0xc0170, 2017-12-01:
The above littlefs resolved this, we should not see the timeouts anymore

Thanks @geky for the travis config improvements

closed 2017-12-01 by 0xc0170