ARMmbed/mbed-os #5588: Plans for SDK 12 or 13 for nrf52832 devices?

jrobeson, 2017-11-26:
### Description
- Type: Question
- Related issue: `#4132`
## Bug

**Target**
Any based on nrf52832

-----------------------------------------------------------------

## Question
  Just wanting to what happened with the effort to use the SDK13 for both nrf52832 devices and nrf52840

jrobeson, 2017-11-26:
I didn't realize that SDK 14 had been out for awhile, so I'm updating the title and description.

0xc0170, 2017-11-27:
cc @ARMmbed/team-nordic

pan-, 2017-11-27:
SDK15 and production version of the Softdevice for the NRF52840 will likely be released soon (NRF52840 production chips expected this quarter). I hope we will get both in a reasonable time frame. However I don't think updates for SDK12 or 13 will be released (maybe 14 if not skipped).

jrobeson, 2017-11-28:
@pan- : so to clarify. It's planned to bring SDK15 into mbed-os for usage with both nrf52832 and nrf52840 devices?

pan-, 2017-11-28:
Nothing is planned yet; I just have hopes.

dlfryar, 2018-01-12:
Mbed OS SDK updates happening [here](https://github.com/ARMmbed/mbed-os/pull/5845)

marcuschangarm, 2018-03-08:
Progress on updating to SDK 14.2 can be followed here: https://os.mbed.com/forum/upcoming-features/topic/29477/

0xc0170, 2018-05-18:
This should be resolved on master now

closed 2018-05-18 by 0xc0170