ARMmbed/mbed-os #5600: No driver present which uses TRNG HAL

MikeDK, 2017-11-28:
### Description

- Type: Enhancement

## Enhancement

**Reason to enhance or problem with existing solution**
There is a well defined HAL API for using TRNG, but there exists no driver which uses it.
The only place where hal/trng_api.h is included is mbedtls.
But if someone only needs true random numbers, and no crypto stuff, the only way now is to use the HAL interface directly.

**Suggested enhancement**
Implement a small driver class with some nice type cast overload operators for...:
- uint8_t
- uint16_t
- uint32_t
- uint64_t
- uint8_t*
- float
- double
- bool (coin flip :) )

... and functions for getting variable lengths...:
- GetString(std::string& str, const uint32_t& len)
- GetArray(std::vector<uint_8>& array, const uint32_t& len)
- GetArray(uint8_t *array, const uint32_t& len)

... maybe also some overloaded shift operators (>>) to do it stream-like.

**Pros**
No need to use HAL directly, and also the possibility for all kinds of type cast overloads.

**Cons**
I guess like for CpuUid, I will be the one implementing it ;-)

0xc0170, 2017-11-28:
@hanno-arm @mazimkhan @sbutcher-arm Please review

sbutcher-arm, 2018-06-01:
Hi @MikeDK,

I'm sorry this issue has been open so long without a serious answer. I'm not quite sure why that is, but will look into it for you.

The idea is that Mbed TLS should provide the RNG for the platform, consuming available entropy sources, such as the TRNG, and outputting a sequence of random numbers through it's own DRBG. The interface for the RNG is [here](https://tls.mbed.org/api/ctr__drbg_8h.html).

That is the design for the system, so this isn't really a bug, although maybe it needs to be better documented - which I'll raise as a separate issue.

Hopefully this answers the question, and so therefore I'm closing the bug.

cc: @RonEld

Patater, 2018-07-20:
@cmonr Could you please close this issue?

@MikeDK Please comment or reopen as you see fit.

 Thanks.

closed 2018-07-20 by cmonr