ARMmbed/mbed-os #5612: Error occurs when export to e2studio by using GR-PEACH board on Online Compiler

TomoYamanaka, 2017-11-29:
### Description
- Type: Question
- Priority: Major 

---------------------------------------------------------------
## Question
When exporting the mbed program to Renesas' IDE **e2studio** by using GR-PEACH, the following error is shown on Online Compiler.

`An error occurred during export. Please contact support`

The situation of this issue is the below.
- If my memory serves correctly, there was no problem until beginning of Oct.
- This error occurs even use Mbed OS 5.4.7(GR-PEACH is active as target device).

<br>
Could you confirm the above?

0xc0170, 2017-11-30:
cc @theotherjimmy

TomoYamanaka, 2017-12-05:
How is going? Any update?

It seems that this problem blocks GR-PEACH users' activity.
Thus, if you could update me on this soon, it would be very helpful.

theotherjimmy, 2017-12-05:
@TomoYamanaka I just verified that the online compiler builds with versions of Mbed OS that support the GR Peach. I'll have to track down why the export is failing.

theotherjimmy, 2017-12-05:
[this line](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/e2studio/__init__.py#L31) is the problem. It's just an undefined variable.

theotherjimmy, 2017-12-05:
2cbca704d8 Was an inconsistent update.

TomoYamanaka, 2017-12-14:
Could you please let me know when the above was merged into Online system? 
I will check whether it is available to export to e2studio.

TomoYamanaka, 2018-01-16:
How is going?
Although I verified this issue by using Mbed OS 5.7.2, it seems that #5657 (is merged in Mbed OS 5.7.1) don't still merge into Online Compiler.

TomoYamanaka, 2018-01-23:
Hi @theotherjimmy 
Please let me know when the above was pushed up the Online Compiler.

toyowata, 2018-02-21:
@theotherjimmy Can you please update the current status?

theotherjimmy, 2018-02-21:
Web team has still not merged the tools update.

toyowata, 2018-03-19:
@theotherjimmy 
The fix for online tools deployed and I confirmed the e2studio exporter works now.

However, I can not export particular program to e2studio.

[Step to reproduce]
* Open the export link below:
https://os.mbed.com/teams/Renesas/code/GR-PEACH_mbed-os-client-ZXingSample/export
* Set [Choose board] GR-PEACH
* Set [Choose toolchain] e2 stidio
* Press [Export] to continue

Then, I got Page Error.

theotherjimmy, 2018-03-19:
@TomoYamanaka Have you checked that the project compiles and does not produce a config error?

TomoYamanaka, 2018-03-20:
In the below program, it seems that the above export error message occur on Online compiler yet. Other programs seem to have no problem regarding to export.
https://os.mbed.com/teams/Renesas/code/GR-PEACH_mbed-os-client-ZXingSample/

export error message:
```
An error occurred during export. Please contact
support@mbed.com.
```

<br>
And, although I checked whether the exported program(ex: mbed-os-example-blinky) compiles on e2studio, I faced many compile error.  
compile error message:

```
fatal error: flash-journal-strategy-sequential/flash_journal_crc.h: No such file or directory
fatal error: flash-journal-strategy-sequential/flash_journal_crc.h: No such file or directory
fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
```

theotherjimmy, 2018-03-21:
@TomoYamanaka I think you'd be hard pressed to find a different online compiler error message. I'll have a look at that reproducing example tomorrow.

theotherjimmy, 2018-03-22:
Oh. That project is too big to export. The online compiler is timing out. I think. @immunda @thegecko Is there something I could do to improve the performance of export so that this does not happen?

thegecko, 2018-03-22:
We could increase the timeout, but perhaps cutting down the export size would be a better way to go? E.g. removing targets and history?

theotherjimmy, 2018-03-22:
@thegecko We already don't include other targets or any history.

thegecko, 2018-03-22:
@theotherjimmy  Please raise an IOTTOO ticket for us to investigate the timeout. A specific re-creatable use case would be really helpful.

theotherjimmy, 2018-03-22:
Let me verify that this example does trigger the timeout. Then I'll raise an issue for it.

theotherjimmy, 2018-03-22:
It's not triggering a timeout. It's a config error that is a bug.

theotherjimmy, 2018-03-22:
I resolved that bug on partnerbeta, but now the requests just die, with no exception or anything.

theotherjimmy, 2018-03-22:
see https://github.com/ARMmbed/mbed-os/pull/6428

toyowata, 2018-04-17:
@theotherjimmy I still got timeout error for the particular project here:
 https://os.mbed.com/teams/Renesas/code/GR-PEACH_mbed-os-client-ZXingSample/export

Can you please check?

yennster, 2018-04-17:
The same error also still occurs with this Renesas project in the online compiler with the GR-PEACH target and e2 Studio IDE: https://os.mbed.com/teams/Renesas/code/GR-PEACH_WebCamera/

MarceloSalazar, 2018-04-24:
Internal reference: IOTMORF-2325

theotherjimmy, 2018-05-21:
@TomoYamanaka I can no longer reproduce this bug.

TomoYamanaka, 2018-06-06:
I confirmed the below.

- **Export to e2studio**
   The phenomenon that I mentioned at first description was fixed, and I could export without error message on Online Compiler.

- **Build on e2studio**
   Unfortunately, I faced the several build error when compiled the exported programs on e2studio.
   The build errors are a bit different by programs, but I frequently face messages such as "No such file or directory".
   Also I can build those programs on online compiler without problem.
   Thus I think that Exporter may have lost some files when exporting.

<br>

program|Export to e2studio|Build on e2studio
------------ | ------------- | -------------
mbed-os-example-blinky|OK|NG
GR-PEACH_mbed-os-client-ZXingSample|OK|NG
GR-PEACH_HVC-P2_sample|OK|NG
GR-PEACH_WebCamera|OK|NG

<br>

For example, when building exported `mbed-os-example-blinky(mbed os 5.8.5)` on e2studio, the following error message is shown.

```
../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.cpp:25:59: fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
 #include "storage-volume-manager/storage_volume_manager.h"
                                                           ^
../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume_manager.cpp:25:59: fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
 #include "storage-volume-manager/storage_volume_manager.h"
                                                           ^
../mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/flash_journal_crc.c:18:65: fatal error: flash-journal-strategy-sequential/flash_journal_crc.h: No such file or directory
 #include "flash-journal-strategy-sequential/flash_journal_crc.h"
                                                                 ^
make: *** [mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume_manager.o] Error 1
mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/subdir.mk:21: recipe for target 'mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume_manager.o' failed
make: *** Waiting for unfinished jobs....
make: *** [mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.o] Error 1
mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/subdir.mk:21: recipe for target 'mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.o' failed
make: *** [mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/flash_journal_crc.o] Error 1
mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/subdir.mk:24: recipe for target 'mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/flash_journal_crc.o' failed
../mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/strategy.c:18:65: fatal error: flash-journal-strategy-sequential/flash_journal_crc.h: No such file or directory
 #include "flash-journal-strategy-sequential/flash_journal_crc.h"
                                                                 ^
make: *** [mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/strategy.o] Error 1
mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/subdir.mk:24: recipe for target 'mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/strategy.o' failed

```

0xc0170, 2018-06-06:
@TomoYamanaka Would it make sense to create a new issue. As this one was fixed - building in the online IDE and also exporting. Now the exporter has an issue. 

> ../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.cpp:25:59: fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
 #include "storage-volume-manager/storage_volume_manager.h"

Looks like exporter also included file from feature storage (but not included headers in the include paths) ?

TomoYamanaka, 2018-06-07:
@0xc0170 
I closed this and created a new issue(#7155 ) since the problem that could not export was fixed.

closed 2018-06-07 by TomoYamanaka