ARMmbed/mbed-os #5615: warnings in code

netanelgonen, 2017-11-29:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
we are trying to compile with treat warning as errors and we are facing errors inside mbed-os
warnings inside the code example:

        [Error] nsdynmemLIB.c@240,25: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
        [Error] nsdynmemLIB.c@382,31: comparison between signed and unsigned integer expressions [-Werror=sign-compare]

- Type: Bug 

- Priority:  Major 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2017-11-29:
@JanneKiiskila please review

netanelgonen, 2017-11-29:
found some more 
        [Error] nsdynmemLIB.c@240,25: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
        [Error] nsdynmemLIB.c@382,31: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
        [Error] nsdynmemLIB.c@240,25: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
        [Error] nsdynmemLIB.c@382,31: comparison between signed and unsigned integer expressions [-Werror=sign-compare]

should i open a new issue? @0xc0170

netanelgonen, 2017-11-29:
found some more warnings:
`[Warning] ppp_lwip.h@58,44: statement with no effect [-Wunused-value]
[Warning] ppp_lwip.h@58,44: statement with no effect [-Wunused-value]
[Warning] coap_security_handler.c@238,12: 'simple_cookie_check' defined but not used [-Wunused-function]
[Warning] coap_security_handler.c@212,12: 'simple_cookie_write' defined but not used [-Wunused-function]
`
@0xc0170 should i open a new issue or this can be handled in this one?

0xc0170, 2017-11-29:
@netanelgonen Every module should have probably own but if we many, you can post them here. 

we would like to eliminate warnings, note, you might find warnings in targets code or elsewhere, some were reported earlier, please check also the issue tracker here. You can find for instance the above lwip warning which comes from the lwip codebase as I can see.

netanelgonen, 2017-11-30:
ok, so i will leave those warnings here...
those are all the warnings i have found, for now.
hopefully this will be enough

netanelgonen, 2017-12-25:
Hi, Any update on those warnings?
this is holding us back from "treat warnings as errors" ourselves

0xc0170, 2018-01-08:
@JanneKiiskila @yogpan01 Who could help to resolve the warnings in common_pal feature?

> coap_security_handler.c@212,12: 'simple_cookie_write' defined but not used [-Wunused-function]

@SeppoTakalo Only used if MBEDTLS_SSL_SRV_C is defined ?

> Warning] ppp_lwip.h@58,44: statement with no effect [-Wunused-value]

@kjbracey-arm Can we fix this warning,

kjbracey-arm, 2018-01-08:
nsdynmemlib warnings were removed by a correction in the master repo in https://github.com/ARMmbed/nanostack-libservice/pull/62

@deepakvenugopal, @arto, can you arrange an update?

kjbracey-arm, 2018-01-08:
Actually, I believe that fix is already merged, via https://github.com/ARMmbed/mbed-os/pull/5511, so you shouldn't be seeing the warning any more in 5.7. That was merged 2 days prior to this issue's original submission - I'm hoping the report was testing old code.

netanelgonen, 2018-01-08:
hi @kjbracey-arm  & @0xc0170 
the original report was not specific to those warnings but for warnings inside mbed-os.
as I'm only working on releases i could not see any merge before a release.

currently, there are still warnings in master (8.1)
this is the latest list of warnings:

[Warning] ip4_addr.h@171,40: the comparison will always evaluate as 'false' for the address of 'ip_addr' will never be NULL [-Waddress]
[Warning] FATFileSystem.cpp@661,0: comparison between signed and unsigned integer expressions [-Wsign-compare]
[Warning] FATFileSystem.cpp@664,0: comparison between signed and unsigned integer expressions [-Wsign-compare]
[Warning] coap_security_handler.c@251,12: 'simple_cookie_check' defined but not used [-Wunused-function]
[Warning] coap_security_handler.c@225,12: 'simple_cookie_write' defined but not used [-Wunused-function]
[Warning] thread_bbr_api.c@693,21: format '%d' expects argument of type 'int', but argument 4 has type 'uint32_t {aka long unsigned int}' [-Wformat=]
[Warning] thread_bbr_api.c@699,21: format '%d' expects argument of type 'int', but argument 4 has type 'uint32_t {aka long unsigned int}' [-Wformat=]
[Warning] protocol_6lowpan_bootstrap.c@2541,5: statement with no effect [-Wunused-value]
[Warning] thread_network_data_storage.c@2454,22: format '%d' expects argument of type 'int', but argument 5 has type 'uint32_t {aka long unsigned int}' [-Wformat=]
[Warning] mpl.c@1133,26: unused variable 'ext' [-Wunused-variable]

0xc0170, 2018-06-21:
Warnings reported were fixed. More warnings were reported in the recent days (and fixed).

Please create a new issue with updated warnings

closed 2018-06-21 by 0xc0170