ARMmbed/mbed-os #5619: error: 4//tmp/tmpFumSHy/mbed_config.h: No such file or directory in import for SW4STM32

Hoel, 2017-11-30:
### Description
- Type: Bug 
- Related issue: `#abc`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
All STM32 tried (F103, L151, F303)

**Toolchain:**
GCC_ARM

**Toolchain version:**
lastest upadtes in online mbed IDE, latest updae in SW4STM32

**Expected behavior**
build without error

**Actual behavior**
error during build  related to mbed_config.h

`14:17:51 **** Incremental Build of configuration Debug for project SX1280PingPong ****
make all 
Building file: ../SX1280Lib/sx1280-hal.cpp
Invoking: MCU G++ Compiler
...
arm-none-eabi-g++: error: 4//tmp/tmpFumSHy/mbed_config.h: No such file or directory
make: *** [SX1280Lib/sx1280-hal.o] Error 1`
 
**Steps to reproduce**
Import example project into online mbed IDE https://os.mbed.com/compiler/#import:/teams/Semtech/code/SX1280PingPong/
compile right clic on project / export program  select SW4STM32. Open project from SW4STM32 (file / open projects from filesystem), build  

The exact same project build correctly when exported in Keil uVision 5

0xc0170, 2017-11-30:
cc @ARMmbed/team-st-mcd

Hoel, 2017-11-30:
after over 4 hours of fixes i got the project compiling fine with default GCC version, there are some serious issues with the export for SW4STM32 from online mbed IDE, i had to remove many linker flags and fix various thing in the code to get it working.

bcostm, 2017-12-01:
Thanks for the debug and information.

`there are some serious issues with the export for SW4STM32 from online mbed IDE`
ST team cannot change the online exporter. Note that the SW4STM32 exporter has been updated recently (PR #4779) and was working fine using mbed CLI.

Hoel, 2017-12-01:
Hi, i can believe SW4STM32 exporter has been updated recently but there are still some issues with the generated project when it is imported in SW4STM32, last version, at least on OSX (i havent checked on Windows). Multiple project have been exported and for each of them some changes had to be made in the project properties to get it compiling. I will detail the steps required later today so that this issue can be fixed in the exporter.

Hoel, 2017-12-16:
sorry for late update, here are the steps required to get a project imported from online compiler to SW4STM32 compiling on OSX:
-go to project properties / C/C++ build / MCU GCC compiler / includes and change includes files to point to the actual mbed_config.h which is at the root of project folder
-go to project properties / C/C++ build / MCU G++ compiler / includes and change includes files to point to the actual mbed_config.h which is at the root of project folder
-go to project properties / C/C++ build / MCU G++ compiler /miscellaneous / linker flags and remove the following flags: -Wl,--wrap=_malloc_r -Wl,--wrap=_free_r -Wl,--wrap=_calloc_r
Some other flags may need to be removed, so far with the limited testing i made these are the only flags i had to remove.

0xc0170, 2017-12-18:
From the report above, seems like config file is not included (preinclude not done?) and wrapping allocators.

@bcostm Can you confirm?

bcostm, 2018-04-26:
We checked the SW4STM32 exporter and it works fine using mbed CLI but there is still an issue when exporting from mbed online IDE. As we cannot do anything on the IDE, can someone in ARM have a look ? Thanks

0xc0170, 2018-04-26:
@theotherjimmy Can you reproduce the above issue report in the online IDE?

ajaakko-arm, 2018-04-27:
Just compared a project exported from the online compiler and a project exported from the command line.
Online compiler:
```
<option id="gnu.c.compiler.option.include.files.656335916" name="Include files (-include)" superClass="gnu.c.compiler.option.include.files" useByScannerDiscovery="false" valueType="includeFiles">
	<listOptionValue builtIn="false" value="${ProjDirPath}//tmp/tmpR29uqI/mbed_config.h"/>
</option>
```
Command line:
```
<option id="gnu.c.compiler.option.include.files.613416024" name="Include files (-include)" superClass="gnu.c.compiler.option.include.files" useByScannerDiscovery="false" valueType="includeFiles">
	<listOptionValue builtIn="false" value="${ProjDirPath}/.\mbed_config.h"/>
</option>
```
Question is, why online compiler adds this temporary path to config header, and should we remove extra path in sw4stm32 exporter? Exporter itself has nothing to do with wrong path:
`
config_header = self.filter_dot(self.toolchain.get_config_header())
`
And I cannot debug online compiler. 
@theotherjimmy : Can you help with this?

theotherjimmy, 2018-04-27:
>Exporter itself has nothing to do with wrong path:
config_header = self.filter_dot(self.toolchain.get_config_header())

Actually, that will always generate the wrong path.

theotherjimmy, 2018-04-27:
You need to do the "subtract basepath" thing on the config header:
```python
config_header = relpath(config_header, resources.file_basepath[config_header])
```

This does nothing offline, and removes the tempdir crud online.

ajaakko-arm, 2018-05-08:
I created pull request, which should fix tmp-path problem. Change was tested to build with mbed-os-example-blinky (both in Windows and Linux). Unfortunately I have no possibility to do testing in OSX.

I did not remove any compiler flags. I think it's better option to fix compiler warnings and errors in SX1280PingPong code, than remove problematic warnings by tuning compiler flags.

PR #6838 (2018-05-14): export: Fix invalid config header path in Sw4STM32

### Description

Removed tmp-path from config header path, when project is exported in online compiler. Fixes #5619 

Parallel build is now enabled by default. In Windows build is terribly slow if parallel build is not enabled.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] New target
    [ ] Feature
    [ ] Breaking change
closed 2018-05-14 by cmonr