ARMmbed/mbed-os #5626: Realtek-RTL8195AM compilation failure for mbed-os-example-blinky

mray19027, 2017-11-30:
Type: Bug
Target: Realtek-RTL8195AM

When compiling on the online compiler, `mbed-os-example-blinky` does not compile with the Realtek-RTL8195AM target.

Error: "/extras/mbed-os.lib/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_ARM_STD/rtl8195a.sct", line 16 (column 22): Error: L6236E: No section matches selector - no section to be FIRST/LAST.

MbedOS version:
Head of 5.6 (Commit https://github.com/ARMmbed/mbed-os/commit/182bbd51bc8d1b6217477c2b5de3f1d0c5e6249f)

0xc0170, 2017-12-01:
To be certain, this is only online environment related? Based on the above, seems like startup file missing in the workspace.

I tested same commit SHA locally, builds. Thus this is related to the online IDE. Please confirm

0xc0170, 2017-12-01:
cc @theotherjimmy

mray19027, 2017-12-04:
@0xc0170 - Can confirm that is only on the online compiler

0xc0170, 2017-12-04:
> @0xc0170 - Can confirm that is only on the online compiler

You can also test it locally to be certain. I did as I wrote above, fetched blinky, same sha for mbed-os and built it without errors.

prashantrar, 2018-01-29:
The PR for this issue has been merged but the online compiler still gives a compilation issue, is the issue still persistent even after merge?

0xc0170, 2018-01-29:
> The PR for this issue has been merged but the online compiler still gives a compilation issue, is the issue still persistent even after merge?

Are you using mbed OS 5.7.3? Please specify what versions and how are you testing

prashantrar, 2018-01-29:
I am using mbed OS 5.7.3, I loaded the example into the compiler using this link 

https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/

0xc0170, 2018-01-29:
> The PR for this issue has been merged but the online compiler still gives a compilation issue, is the issue still persistent even after merge?

Please also specify what errors are you getting, more information you can provide, better response you can get.

@ARMmbed/team-realtek Can you reproduce?

I tested it just now, getting internal compiler error for this target (5.7.3 mbed-os, blinky example), other targets build fine (freedom k64f, nucleo f429

prashantrar, 2018-01-29:
[image]

This is the snapshot of the issue that im getting

prashantrar, 2018-02-06:
Is there any way we can debug the root cause of this issue from our side?

samchuarm, 2018-02-12:
Hi @0xc0170 @theotherjimmy  any other guidance we could share with Realtek to debug this issue from their side? Thanks

samchuarm, 2018-03-02:
#6216

0xc0170, 2018-03-28:
Should be resolved via #6216

closed 2018-03-28 by 0xc0170