ARMmbed/mbed-os #5632: lpc546xx: ISP able pins as gpio

gorazdko, 2017-12-01:
## Type: Question

## Abstract

I configured two pins as outputs - one of them supports ISP mode (P0_13):
```
DigitalOut p1(P0_13);
DigitalOut p4(P0_16);
```

[image]

Running this code works fine. Both pins are toggling:
```
 while (true) {						
    p1.write(0);
    p4.write(0);	  
    wait(0.5);	

    p1.write(1);
    p4.write(1);

    wait(0.5);			
    }
```

Though my debugger is not showing any toggling for P0_13, it cannot be manipulated over debugger nor read correctly (always 0)

[image]

Nor can the value be read correctly for P0_13 in code:   `p1.read()`.  *Is this behaviour intentional? Or are there any other registers that need to be configured to be able to read the value correctly?*

I am asking because programs that use operator overloading (which use gpio read method) dont work correctly for ISP able pins:

```
 while (true) {						
        p1 = !p1;      // ISP able;   does not work beacuse the value cannot be read correctly
	p4 = !p4;     // ISP unable, works fine 
        wait(0.5);			
    }
```

*Why are MODE and OD bits missing in PIO013 (first image)?*

### Remark
All ISP mode pins are always pulled up

gorazdko, 2017-12-01:
@mmahadevan108 could you comment on this?

mmahadevan108, 2018-01-12:
I don't see the issue. I tried the below test:

#include "mbed.h"

int main() {
    DigitalOut p1(P0_13);
    DigitalOut p4(P0_16);

    while (true) {
        int i;
        p1.write(0);
        p4.write(0);
        wait(0.5);

        i = p1.read();
        printf("After clear I=%d\n", i);

        p1.write(1);
        p4.write(1);
        
        wait(0.5);
        i = p1.read();
        printf("After set I=%d\n", i);
    }

    return 0;
}

Output was as expected:
    After clear I=0
    After set I=1

mmahadevan108, 2018-01-12:
`uykjhkjhk`

closed 2018-07-04 by gorazdko