ARMmbed/mbed-os #5636: Booting from SD possible?

amq, 2017-12-01:
### Description
- Type: Question

---------------------------------------------------------------

## Question

Is it possible to start the application from a binary on an SD card? Naturally, I would have a bootloader in flash.

deepikabhavnani, 2017-12-01:
In order to load application from SD Card, bootloader should have support to read data from SD card. If bootloader supports reads from SD card and load application in RAM then yes it should be feasible.

0xc0170, 2017-12-07:
@amq Does this answer your question above?

amq, 2017-12-07:
@0xc0170 not really. I was looking for at least a short sketch of how loading the application should work, with the assumption that all requirements like SD and bootloader are met.

deepikabhavnani, 2017-12-07:
@amq - I don't think all requirements are met in current bootloader and mbed-os. We might think of that in future as enhancement.

closed 2017-12-12 by amq