ARMmbed/mbed-os #5638: make _sbrk() weak (for malloc() from external memory)

bmcdonnell-ionx, 2017-12-01:
### Description
- Type: Bug | Enhancement
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
LPC40xx

**Toolchain:**
GCC_ARM

**mbed-os sha:**
(182bbd51b)

----------------------------------------------------------------
## Enhancement

**Suggested enhancement**

In the file `mbed-os/platform/mbed_retarget.cpp`, please add the attribute `MBED_WEAK` (or equivalent) to the function definition `_sbrk()`, to allow the user to override it.

```
extern "C" MBED_WEAK caddr_t _sbrk(int incr) {
    // [snip...]
}
```

**Reason to enhance or problem with existing solution**

An approach sometimes used to access external memory in embedded systems is to have all stack memory use internal RAM, and all dynamic memory (e.g. malloc()) allocate memory from external RAM. This is especially useful during development, to at least temporarily skip over the often difficult task of editing linker files.

Embedded Artists' [EALib](https://os.mbed.com/users/embeddedartists/code/EALib/file/e1e36493f347/sdram.cpp/) takes this approach to enable access to an external SDRAM chip. In sdram.cpp, it defines/overrides the `_sbrk()` function, which is used by `malloc()`, (etc.), to manage the heap.

This worked in older versions of mbed (at least through rev 120), but I cannot build it in the latest mbed-os, due to `_sbrk()` already being defined in `mbed_retarget.cpp`. Please make the mbed-os `_sbrk()` weak, to allow EALib to use the SDRAM as intended.

**Pros**

Described above.

**Cons**

Problematic if another library also defines `_sbrk()` weakly.

0xc0170, 2017-12-04:
@c1728p9 @SenRamakri

bmcdonnell-ionx, 2017-12-06:
#1819 was trying to do the same thing for a different platform, but it takes a different approach. I don't think it's the right approach here, because it's making mbed-os *always* use the external memory for dynamic memory for that platform, whereas I want to give the user the option to do so.

But I give details here for reference.

-----

Upon closing the issue, @ccli8 [said](https://github.com/ARMmbed/mbed-os/issues/1819#issuecomment-266331660):

> I overwrite _sbrk with conditional compile in retarget.cpp. This modification has merged and so I close the issue.

I think he's referring to these commits, in particular the changes to platform/retarget.cpp:
[b954780](https://github.com/ARMmbed/mbed-os/commit/b95478015bec3e31d100bfc1fe6b865fdc9aa964#diff-e08d01c8b7bf369bbb3d5d70788a4b88)
[dfc3240](https://github.com/ARMmbed/mbed-os/commit/dfc32409a96f7927472b5085a565ccc18d805531#diff-e08d01c8b7bf369bbb3d5d70788a4b88)

bmcdonnell-ionx, 2018-01-26:
I may do a PR for this soon.