ARMmbed/mbed-os #5639: UART buffer size not accesible in constructor

sarahmarshy, 2017-12-01:
### Description
- Type: Enhancement

Inspired by - https://os.mbed.com/questions/75881/Asynchronous-serial-read-buffer-size/

The current case hides the buffer size from the user using the config system - https://github.com/ARMmbed/mbed-os/blob/master/drivers/UARTSerial.h#L32-L34. Users should be able to specify it in the constructor of the serial object. The default size should be 256. Like so:
```
UARTSerial(PinName tx, PinName rx, int baud = MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE, int buff_size = 256);
```

It looks like the buffers are declared statically now -https://github.com/ARMmbed/mbed-os/blob/master/drivers/UARTSerial.h#L235-L236. So, the work will require dynamically allocated these buffers.

@kjbracey-arm

kjbracey-arm, 2017-12-04:
Yes, the static buffer sizes arise because it uses the `CircularBuffer` template. There's no dynamic-sized equivalent to that in the system. I'd also like to have this be dynamic, but I do know @sg- is keen on reducing dynamic allocation. (Although does a pre-main construction alloc in a static object constructor count as dynamic?)

I imagine the buffer sizes could be added as a template parameter, but not sure if that's too weird. Is there a dynamic-size enhancement request for `CircularBuffer` already filed?

However, in most cases people try to increase the UARTSerial/BufferedSerial RX buffer size, it isn't the correct fix for whatever problem they're having.

256 bytes of driver buffer works well in most cases. A 2048-byte application buffer in conjunction with a 256-byte driver buffer may be quite appropriate.

Eg for a AT-based network driver - you issue a command, then `ATCmdParser` start reading the response lines into its 2048-byte line buffer. The 256-byte driver buffer covers delays in it getting around to starting the read, or it being pre-empted by other threads. It **isn't** intended to hold the whole of one AT response, and doesn't need to as responses are read immediately after issuing the AT command. 

For higher-speed data streaming use like PPP, you have a dedicated data pump whose job it is to transfer stuff between the serial and the network stack. As that pump's continuously functioning, there should never be time for 256 bytes of driver buffer to fill, even while transferring 1500-byte frames.

I _can_ imagine a design where you configure and rely on a big driver buffer, but that should be a substitute for having a big (or any) application buffer. Having both buffers be big would be inefficient design - eg letting 2K of data accumulate in the driver buffer, then reading it all into a 2K application buffer, so needing 4K of RAM. 

Summary - yes, it would be nice, but if this was available, I suspect the main result would be unnecessarily oversized buffers covering up design problems.

It would be a very easy beginner error to assume "I'm reading into a 2K application buffer, so I should configure 2K of driver buffer to match".

The 256-byte buffer is intended to be a drastic improvement over the 0 in `Serial`, allowing apps that read promptly from threads to be reliable without flow control. But just because there is some buffer in there, it doesn't generally make it appropriate to make it big to let apps put off reading for prolonged periods.

sg-, 2017-12-04:
> I'd also like to have this be dynamic, but I do know @sg- is keen on reducing dynamic allocation. (Although does a pre-main construction alloc in a static object constructor count as dynamic?)

Nope :) If new is exclusive to the constructor and delete happens to be missing I think that would work. In general, I'd like to see warning messages when delete and free are used in a debug profile for as a way to get a handle on what's actually freeing memory.

closed 2019-10-31 by ciarmcom