ARMmbed/mbed-os #5641: Is there an FAQ or TL;DR for bridging various SDK functionalities to mbed  OS?

jrobeson, 2017-12-03:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question

## Question

I was trying to setup I2S on an NRF52 and mbed provides no nice driver wrapper for it. I was having all sorts of trouble with my application hanging. I was eventually able to figure out that the interrupt vector table was remapped and that all I needed was to grab the I2S drivers from the Nordic SDK (11) wrap it in a `extern "C"` (already done in SDK 13) and add `NVIC_SetVector(I2S_IRQn, (uint32_t) I2S_IRQHandler);`  

Is there a list of gotchas like this that would help folks who are just playing around with embedded dev and trying to get stuff to work?

dlfryar, 2018-01-12:
Mbed OS SDK updates happening [here](https://github.com/ARMmbed/mbed-os/pull/5845)

jrobeson, 2018-01-13:
that PR doesn't really cover my question.

0xc0170, 2018-01-16:
I am not certain what you are after in the documentation. I am not aware of any list like this, as SDKs are  different.  NVIC Set/Get vector API should be documented (it shall be covered or at least referenced in the porting guide - how to register interrupt handler). And this API is defined by CMSIS