ARMmbed/mbed-os #5642: REALTEK_RTL8195AM - LEDs broken/not mapped

JanneKiiskila, 2017-12-03:
### Description
- Type: Bug
- Related Issue: `#abc`

---------------------------------------------------------------
## Bug

**Version information**
(`mbed ls`)

```
mbed-os-example-blinky (5e5c7ad85122)
`- mbed-os (2e1c2a1cdf16)
```

**Environment details**
Describe against which environment you are testing.
Compiler & other relevant tools - version information.

ARM_GCC comiing with MBed CLI 0.43 version.

**Expected Behavior**

LED would blink.

**Actual Behavior**

No blinking.

**Steps to Reproduce**

1. `git clone git@github.com:ARMmbed/mbed-os-example-blinky.git`
1.  `mbed compile -m REALTEK_RTL8195AM -t GCC_ARM`
1. flash board (depending on your OS choice)

Is the LED1 definition broken somehow?  The board definitely has multiple LEDs.

@Archcady  @tung7970

Actually, the TARGET file explains a lot;

```
TARGET_AMEBA/TARGET_RTL8195A/PinNames.h:    /* LED1~4 are defined as alias of GPIO pins, they are not the LEDs on board*/
TARGET_AMEBA/TARGET_RTL8195A/PinNames.h:    LED1        = PB_4,
TARGET_AMEBA/TARGET_RTL8195A/PinNames.h:    LED2        = PB_5,
TARGET_AMEBA/TARGET_RTL8195A/PinNames.h:    LED3        = PB_6,
TARGET_AMEBA/TARGET_RTL8195A/PinNames.h:    LED4        = PB_7,
```

Is there any special reason why none of the board LEDs can't be used?

JanneKiiskila, 2017-12-04:
Interestingly enough, if you add prints to mbed-os-example-blinky, the board will then blink. Seems the LED D5 is mapped somehow to serial?

samchuarm, 2017-12-04:
@ARMmbed/team-realtek

v2422, 2017-12-04:
Hi @JanneKiiskila,

The LEDs on board are used by CMSIS-DAP as well as power indicator. There is no LED connected to RTL8195A chip.

samchuarm, 2017-12-04:
[https://github.com/ARMmbed/mbed-os-example-blinky/issues/92](url)

samchuarm, 2017-12-04:
#https://github.com/ARMmbed/mbed-os-example-blinky/issues/92

0xc0170, 2017-12-04:
Should this be documented - https://os.mbed.com/platforms/Realtek-RTL8195AM/ ? I assumed this was already captured as known-limitation of the board.

Archcady, 2017-12-04:
There are no LED's conncted to AMEBA , they are all used by the DAP only and are not user programmable. I shall update this on the platform page. Usually we test the arduino blinky example by connecting an external LED, the same can be updated for the mbed-os-blinky example. Any other suggestions here?

Archcady, 2017-12-04:
Updated platform page under the "warning section"

JanneKiiskila, 2017-12-04:
I think the mbed-os-example-blinky docs should be updated, too.
Need to tweak my mbed-os-example-client PR as well to highlight this.

Archcady, 2017-12-04:
Is there any need for modifying pinnames.h?

JanneKiiskila, 2017-12-04:
I don't think, we just need to now document how you have tested the blinky, i.e. to which pins do you connect the LED.

v2422, 2017-12-06:
Please connect LED with GPIOB_4 (INT/PWM0) and GND. The pin can be found in the diagram of "RTL8195AM pinout (right)" in https://os.mbed.com/platforms/Realtek-RTL8195AM/

Archcady, 2017-12-14:
I have updated the platform page with an Example on how to connect LED to the RTL8195AM

0xc0170, 2017-12-14:
>I have updated the platform page with an Example on how to connect LED to the RTL8195AM
 
This is now resolved

closed 2017-12-14 by 0xc0170