ARMmbed/mbed-os #5643: REALTEK_RTL8195AM - DAPLINK needs updating

JanneKiiskila, 2017-12-03:
### Description
- Type: Enhancement 
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**DAPLink version:**

```
mbed detect
No handlers could be found for logger "mbedls.platform_database"
WARNING:mbedls.lstools_base:Call to deprecated function get_details_txt. This method will be removed from the public API. Please use 'list_mbeds' instead

[mbed] Detected REALTEK_RTL8195AM, port COM5, mounted D:, interface version 0241:
[mbed] Supported toolchains for REALTEK_RTL8195AM
+-------------------+-----------+-----------+-----------+-----------+-----------+-----------+
| Target            | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |   ARMC6   |
+-------------------+-----------+-----------+-----------+-----------+-----------+-----------+
| REALTEK_RTL8195AM |     -     | Supported | Supported | Supported | Supported | Supported |
+-------------------+-----------+-----------+-----------+-----------+-----------+-----------+
Supported targets: 1
Supported toolchains: 4
```

**Expected behavior**

We should use DAPLINK 244, at least.
DAPLINK's newer versions bring some important USB stability fixes that are very useful in CI / automated test environments.

**Actual behavior**

2.41 used.

**Steps to reproduce**

Shown above.

samchuarm, 2017-12-04:
@ARMmbed/team-realtek

0xc0170, 2017-12-04:
@JanneKiiskila To be clear, this is for realtek to update their interface implementation to used later DAPlink version (at first I was not sure if this is for our CI to get an update).

JanneKiiskila, 2017-12-04:
Yes, Realtek should update the daplink (aka `firmware.bin`) they offer via the links in official mbed os board page (https://os.mbed.com/platforms/Realtek-RTL8195AM/).

JanneKiiskila, 2018-01-10:
@Archcady @tung7970  - any chance of getting this one sorted out?

JanneKiiskila, 2018-01-29:
Sure!

closed 2018-01-29 by JanneKiiskila