ARMmbed/mbed-os #5645: UBLOX_EVK_ODIN_W2 does not pass Greentea test-case WIFI_SET_CREDENTIAL

jarlamsa, 2017-12-04:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-g++.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

**mbed-cli version:** 
`1.2.2`

**mbed-os sha:**
`2e1c2a1cd`

**DAPLink version:**
`0221`

**Expected behavior**
WIFI_SET_CREDENTIAL succeeds

**Actual behavior**
WIFI_SET_CREDENTIAL fails when giving a NULL SSID.
Expected return value is NSAPI_ERROR_PARAMETER
    [1512374386.53][CONN][RXD] >>> Running case #1: 'WIFI-SET-CREDENTIAL'...
    [1512374386.58][CONN][INF] found KV pair in stream: {{__testcase_start;WIFI-SET-CREDENTIAL}},  queued...
    [1512374387.63][CONN][RXD] :32::FAIL: Expression Evaluated To FALSE
    [1512374387.68][CONN][INF] found KV pair in stream: {{__testcase_finish;WIFI-SET-CREDENTIAL;0;1}}, queued...
    [1512374387.77][CONN][RXD] >>> 'WIFI-SET-CREDENTIAL': 0 passed, 1 failed with reason 'Assertion Failed'
    [1512374387.77][CONN][RXD]
    [1512374387.83][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'

**Steps to reproduce**
    mbed test --compile -c -t GCC_ARM -m UBLOX_EVK_ODIN_W2 --app-config .\TESTS\network\wifi\template_mbed_app.txt -n tests-network-wifi
    mbedhtrun -f .\BUILD\tests\UBLOX_EVK_ODIN_W2\GCC_ARM\TESTS\network\wifi\wifi.bin -d X: -p COMY:9600

jarlamsa, 2017-12-04:
@VeijoPesonen @SeppoTakalo @kjbracey-arm @KariHaapalehto @juhaylinen

0xc0170, 2017-12-04:
cc @ARMmbed/team-ublox

andreaslarssonublox, 2017-12-04:
This will be fixed in the next release of the driver.

TuomoHautamaki, 2018-03-21:
Any estimate when the next release will happen?

juhaylinen, 2018-09-04:
Seems to be fixed in https://github.com/ARMmbed/mbed-os/pull/7579

screamerbg, 2018-09-19:
Great. This issue should be closed once 5.10.1 is out.

closed 2018-10-02 by adbridge