ARMmbed/mbed-os #5647: Proper value for SCB_ICSR_RESERVED_BITS_MASK

jrobeson, 2017-12-04:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug?
- Related issue: `#4927`

---------------------------------------------------------------
## Bug

**Target**
any Nordic NRF52 device? maybe NRF51?

**Toolchain:**
GCC_ARM

**mbed-os sha:**
(`git log -n1 --oneline`)

In #4927 it was mentioned by @pan- and @nvlsianpu that the value of `SCB_ICSR_RESERVED_BITS_MASK` should be something else. Likely `0x94403F03F`.

I'm opening this as an issue in case it got forgotten due to being in a closed ticket.

0xc0170, 2017-12-04:
@ARMmbed/team-nordic

nvlsianpu, 2017-12-04:
Known issue for me - @jrobeson Thx for describing the issue.

jrobeson, 2018-03-11:
@marcuschangarm: do you have any insights here?

marcuschangarm, 2018-03-12:
We could change the fixed value to a collection of bit masks using the macros defined here: https://github.com/ARMmbed/mbed-os/blob/master/cmsis/TARGET_CORTEX_M/core_cm4.h#L481-L510

However, I'm not sure that code is even valid anymore. From the `nrf_pwr_mgmt.c` file in `SDK 14.2`:

```
void nrf_pwr_mgmt_run(void)
{
    PWR_MGMT_FPU_SLEEP_PREPARE();
    PWR_MGMT_SLEEP_LOCK_ACQUIRE();
    PWR_MGMT_CPU_USAGE_MONITOR_SECTION_ENTER();
    PWR_MGMT_DEBUG_PIN_SET();

    // Wait for an event.
#ifdef SOFTDEVICE_PRESENT
    if (nrf_sdh_is_enabled())
    {
        ret_code_t ret_code = sd_app_evt_wait();
        ASSERT((ret_code == NRF_SUCCESS) || (ret_code == NRF_ERROR_SOFTDEVICE_NOT_ENABLED));
        UNUSED_VARIABLE(ret_code);
    }
    else
#endif // SOFTDEVICE_PRESENT
    {
        // Wait for an event.
        __WFE();
        // Clear the internal event register.
        __SEV();
        __WFE();
    }

    PWR_MGMT_DEBUG_PIN_CLEAR();
    PWR_MGMT_CPU_USAGE_MONITOR_SECTION_EXIT();
    PWR_MGMT_SLEEP_LOCK_RELEASE();
}
```

jrobeson, 2018-06-19:
@marcuschangarm : so what exactly do you mean by that?  It seems like nobody else has raised this as an issue, so maybe it's just fine as it is? I know I don't know.

marcuschangarm, 2018-06-19:
I don't think 5.9 (and SDK 14.2) has been out long enough for people to notice whether the power consumption is exactly as expected. 

I'm waiting for some better power measurement equipment myself.

TacoGrandeTX, 2018-09-12:
The current value of `SCB_ICSR_RESERVED_BITS_MASK` is not restrictive enough and will prevent the core from entering WFE if any interrupt is pending, even if it is not enabled.  This can happen rather easily as one might lazily disable a low priority interrupt and not notice that the pending bit was set.  This will forever prevent the core from sleeping.  In fact this goes even deeper as architecturally an interrupt can go pending even if it is disabled in the NVIC.

We should just be checking against the VECTPENDING and VECTACTIVE bit fields in the Interrupt Control and State Register. When VECTPENDING is non-zero it indicates which interrupt is pending and enabled.  Presumably we don't want to enter WFE in an exception handler, so we should also include VECTACTIVE in the bit mask.

TacoGrandeTX, 2018-09-12:
As Marcus points out this is only relevant if the SoftDevice is disabled:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/sleep.c#L53-L86

deepikabhavnani, 2018-12-28:
> In #4927 it was mentioned by @pan- and @nvlsianpu that the value of SCB_ICSR_RESERVED_BITS_MASK should be something else. Likely 0x94403F03F.

0x9 4403 F03F. - Seems incorrect to me. Current value is 0x9E43F03F (https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/sleep.c#L32)

Is it proposed to change from 0x9E43 F03F to 0x9443 F03F? 

31: NMIPENDSET - 1 - Correct 
28: PENDSVSET - 1 - Correct
27: PENDSVCLR -  1 (Was) - Should be 0 Set and Clear both cannot be true
26: PENDSTSET - 1(Was) - Correct
25: PENDSTCLR - 1(Was) - Should be 0. Set and Clear both cannot be true.

kjbracey-arm, 2019-01-02:
This was discussed extensively in #8366 - the state at end of discussion there was consensus that the overall implementation is unreliable and not as intended, and messing with the mask value doesn't get to the root of the issue. I'd really rather pick up from there, than go back and just adjust the mask.

I could prepare a PR based on how we were there, but I'm not 100% clear on the various softdevice present/enabled/whatever combos are possible. I need to get in touch with someone who properly understands that layer to determine exactly what is required.

cmonr, 2019-01-03:
It sounds like the way to progress is found in @TacoGrandeTX's comment: https://github.com/ARMmbed/mbed-os/pull/9211#issuecomment-450816674