ARMmbed/mbed-os #5652: LittleFS mkdir error code for existing directory returns LFS_ERR_INVAL

itayzafrir, 2017-12-04:
### LittleFS mkdir error code for existing directory returns LFS_ERR_INVAL
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Expected behavior**
mkdir for an existing directory to return an error code which describes the error

**Actual behavior**
mkdir for an existing directory returns LFS_ERR_INVAL

**Steps to reproduce**
create the same directory twice

@geky

geky, 2017-12-04:
Hi @itayzafrir, thanks for reporting this issue.

Do you have a code snippet that reproduces this problem? I'm having a hard time reproducing.

Currently mkdir on existing directory is tested on all three APIs ([here](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/littlefs/littlefs/tests/test_dirs.sh#L59), [here](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/littlefs/TESTS/filesystem/dirs/main.cpp#L210-L211), and [here](https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/littlefs/TESTS/filesystem_retarget/dirs/main.cpp#L210-L211)), so the issue you're seeing might not be straightforward.

itayzafrir, 2017-12-05:
Sorry my bad, this is not an issue.

closed 2017-12-05 by itayzafrir