ARMmbed/mbed-os #5653: remove non empty directory returns inconsistent error codes between LittleFS & FatFS

itayzafrir, 2017-12-05:
### Attempting to remove/delete a non-empty directory from the FS returns different error codes when using FatFS vs LittleFS

- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Expected behavior**
The same error code will be returned for both file systems

**Actual behavior**
When trying to remove a non-empty directory using FatFS: error code = EACCES
When trying to remove a non-empty directory using LittleFS: error code = EINVAL

**Steps to reproduce**
remove a non empty directory using FatFS and then using LittleFS

@geky

geky, 2017-12-05:
Thanks for reporting this issue. According to [Open Group](http://pubs.opengroup.org/onlinepubs/009695399/functions/rmdir.html), this should be ENOTEMPTY.

geky, 2018-04-29:
Ok, I added a patch to https://github.com/ARMmbed/mbed-os/pull/6336 that should take care of this for FAT. LittleFS was updated in https://github.com/ARMmbed/mbed-os/pull/6179 to use ENOTEMPTY.

So this looks like it should be solved soon, let me know if you have any concerns.

0xc0170, 2018-06-21:
Thanks @geky. This should be resolved in 5.9.

closed 2018-06-21 by 0xc0170