ARMmbed/mbed-os #5658: Is DEVICE_STDIO_MESSAGES well implemented ?

jeromecoutant, 2017-12-05:
Hi

I didn't find in handbook what the behavior should be for DEVICE_STDIO_MESSAGES.

Is it ?
- if DEVICE_STDIO_MESSAGES is defined, then use STDIO_UART_TX and STDIO_UART_RX for default printf
- else, printf becomes a NULL macro

So maybe some code is missing:

in https://github.com/ARMmbed/mbed-os/blob/2e1c2a1cdf168dd739e777165b21f6cd0509c188/platform/mbed_retarget.cpp#L120

I would expect:
#if DEVICE_SERIAL && DEVICE_STDIO_MESSAGES

0xc0170, 2017-12-06:
`DEVICE_STDIO_MESSAGES` -  this needs a look. I was also confused about `DEVICE_STDIO_MESSAGES` as being in `device_has`. This is more optimization for a target, rather than a target HAL attribute.
I would expect this to be a rather configuration option.

> So maybe some code is missing:

To the line 120, it is correct as it is , it's not using stdio, but HAL API, thus the condition there is correct as it is. All it needs its serial HAL implementation.

jeromecoutant, 2018-09-10:
Hi
Is it possible to keep SERIAL feature enabled,
and remove the default STDIO configuration ?
(replace default console by null ?)

kjbracey-arm, 2018-09-11:
If a target wants a null console, just provide an implementation of `mbed::mbed_target_override_console`, returning a dummy `FileHandle` that does nothing.

`Sink` inside `mbed_retarget.cpp` (which is used by default when `DEVICE_SERIAL` is off) kind of does this, except it has weird "endless stream of 0s" input, for backwards compatibility. Borrow/adapt that.

jeromecoutant, 2019-01-02:
Hi
I don't have really any specific issue on my side :-)
Maybe this question should be closed here, and opened in the doc side ?
(STDIO_MESSAGES device_has description is missing)

closed 2019-10-24 by 0xc0170