ARMmbed/mbed-os #5659: DISCO_L475VG_IOT01A exporters work offline but not in online compiler

BlackstoneEngineering, 2017-12-05:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
DISCO_L475VG_IOT01A

**Toolchain:**
All

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
1394bf95f Merge pull request #5474 from sarahmarshy/sigio-test-fix

**Expected behavior**
Expecting online exporters to exist and work. 

**Actual behavior**
There are no functional exporters for this target from the online compiler. 
[image]

There are however a number of exporters from mbed-cli offline

```
| Platform              | atmelstudio | coide | ds5_5 | e2studio | eclipse_armc5 | eclipse_gcc_arm | eclipse_iar | embitz | gcc_arm | gnuarmeclipse | iar | kds | lpcxpresso | make_armc5 | make_armc6 | make_gcc_arm | make_iar | mcuxpresso | qtcreator | simplicityv3 | sw4stm32 | uvision | uvision5 | vscode_armc5 | vscode_gcc_arm | vscode_iar |
| DISCO_L475VG_IOT01A   |      -      |   -   |   -   |    x     |       x       |        x        |      x      |   x    |    x    |       x       |  -  |  -  |     -      |     x      |     -      |      x       |    x     |     -      |     x     |      -       |    x     |    x    |    x     |      x       |       x        |     x      |
```

0xc0170, 2017-12-06:
@theotherjimmy please review this issue

theotherjimmy, 2017-12-06:
This is an IDE frontend issue. check https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/export

0xc0170, 2017-12-07:
> This is an IDE frontend issue. check https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/export

@BlackstoneEngineering Can you report it to the IDE frontend team please?

BlackstoneEngineering, 2017-12-07:
@0xc0170 who is the IDE Frontend team now? Web team or tools team?

thegecko, 2017-12-07:
Found the issue (a case sensitivity mismatch). Will create a fix.

thegecko, 2017-12-07:
This issue duplicates https://github.com/ARMmbed/mbed-compiler/issues/23

closed 2017-12-07 by thegecko

tejashah88, 2017-12-08:
@thegecko The issue that you're referring to leads to a 404 error for me.

thegecko, 2017-12-09:
It's a private repository

thegecko, 2017-12-13:
This is now deployed