ARMmbed/mbed-os #5668: STM32: Compiler does not include code for CAN_2 and CAN_3 with mbed-cli and GCC_ARM

adamheinrich, 2017-12-06:
File [can_api.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/can_api.c) uses the following mechanism to include code for CAN2/CAN3:

```
#if defined(CAN3_BASE) && defined(CAN_3)
/* Code goes here */
#endif
```

This work perfectly for `CAN3_BASE` (which is a macro defined in the STM32 Cube HAL) but does not work with `CAN_3` (which member of an enum defined in target-specific `PeripheralNames.h`). This is probably because GCC doesn't consider enum members as "defined".

A quick hack to make this work under `GCC_ARM` is to manually add `-DCAN_3=CAN_3` to `mbed compile`, which is not very convenient.

A correct solution to rewrite the mechanism or look if there is a GCC flag (and possibly a flag for other compilers as well) which would make the `defined()` macro recognize enum entries.

cmonr, 2017-12-06:
Hi @adamheinrich,

I take it your PR https://github.com/ARMmbed/mbed-os/pull/5486 is meant to fix this issue?

Also, would it be possible to retest with mbed-os-5.7.0 once it's released? It looks like a separate PR (https://github.com/ARMmbed/mbed-os/pull/5486) will already address the issue.

adamheinrich, 2017-12-07:
Hi @cmonr,
no, #5667 fixes data reception on CAN3. This bug is a separate issue which (probably) only occurs with a specific compiler because the particular construction in the code is not compiler-agnostic.

Yes, #5486 is related to the issue because it introduced `defined(CAN_2)` and `defined(CAN_3)`.

closed 2017-12-27 by adamheinrich