ARMmbed/mbed-os #5671: I2C stops working after some time (/SPI collision?)

shayo, 2017-12-07:
- Type: Bug 
- Priority:  Major 
---------------------------------------------------------------
## Bug
I recently upgraded to version 5.6 and now I am experiencing some weird behavior.
The app runs fine, but after some (variable?) time, I2C stops working and all read operations time out (and it takes 10 seconds by default?)
I noticed this because suddenly the entire app became very slow. I then traced it to interfacing with peripherals that are I2C.
The circuit has both I2C and SPI, so I wonder if this is related to some old problem that was present a couple of months ago.

Everything works great with a much older version. 
I don't have a small demo to demonstrate it, but I thought I'll open this and see if this may ring a bell due to some recent changes you introduced...

**Target**
NRF52832

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
1.1.1

**Steps to reproduce**
Not yet...

0xc0170, 2017-12-07:
@shayo Hi, can you provide at least versions that were functional and the one that does not work? That we could use to compare ? You can check the diff between those, mainly regarding SPI and I2C, if there were any changes in HAL that could lead to this.

It would be helpful to get more details, at least to localize changes.

> I don't have a small demo to demonstrate it, but I thought I'll open this and see if this may ring a bell due to some recent changes you introduced...

@ARMmbed/team-nordic were there changes that could lead to this behavior ?

shayo, 2017-12-07:
I upgraded from version 123, so there were many changes to i2c.c
It seems to be localized to the i2c library. when I replaced the files of version 156 with the ones of 123 things seem to be working fine now.

MarceloSalazar, 2018-01-11:
FYI @NeilMacMullen

NeilMacMullen, 2018-01-12:
@shayo  - I assume you are already aware of the restriction at https://github.com/ARMmbed/mbed-os/issues/5843 ?   Basically, you cannot run both I2C and SPI simultaneously from different threads. :-( 

We have also seen an issue where the TWI appears to be generating incorrect waveforms on the I2C bus.  ARM are aware of this and discussing with Nordic I believe.

shayo, 2018-01-14:
This might be the case. I wasn't aware of this limitation.

MarceloSalazar, 2018-04-23:
Enhancement for nrf52 is available on master and should fix this: https://github.com/ARMmbed/mbed-os/pull/6547
Please re-open if the problem continues.

closed 2018-04-23 by MarceloSalazar