ARMmbed/mbed-os #5672: REALTEK-RTL8195AM - BUTTON1/2 definition missing

JanneKiiskila, 2017-12-07:
### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**Toolchain:**
N/A

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)
1.22

**mbed-os sha:**
(`git log -n1 --oneline`)

mbed os 5.7 RC1

**DAPLink version:**
N/A

**Expected behavior**

BUTTON1/BUTTON2 defines found in target defines.
The hardware has 2 buttons next to the USB-connectors that ought to be mapped to the device? 

**Actual behavior**

BUTTON1/BUTTON2 are not defined.

**Steps to reproduce**

0xc0170, 2017-12-07:
@ARMmbed/team-realtek

JanneKiiskila, 2017-12-07:
I am starting to have a feeling these buttons are also connected to the DAPLINK host CPU, instead of the Realtek CPU. Might be they are not thus usable for apps, but I would like to get a confirmation. If that's the case - we should update the board page.

prashantrar, 2017-12-07:
Hello @JanneKiiskila the buttons in the bottom are for DAP while the one on the side is for reset. The buttons arent user programmable.

JanneKiiskila, 2017-12-07:
It's a documentation issue then, raising IOTWEB ticket.

samchuarm, 2017-12-07:
@Archcady please update RTL8195AM platform page for these buttons accordingly. Thanks.

Archcady, 2017-12-14:
Have updated platform page with this detail, also added a link to our official Realtek landing page that shows how to test button by connecting an external button

0xc0170, 2017-12-14:
I can confirm the above, will close this issue as resolved

closed 2017-12-14 by 0xc0170