ARMmbed/mbed-os #5676: make.py generated duplicate file entries in linker list

jorisa, 2017-12-08:
### Description
- Type: Bug (Question?)
- Priority: Minor

## Bug

**Target**
NUCLEO_F767ZI (not relevant?)

**Toolchain:**
GCC_ARM

**Toolchain version:**
gcc-arm-none-eabi-6-2017-q2-update

**mbed-os sha:**
2e1c2a1

**Expected behavior**
`.link_files.txt` should never have duplicate file entries

**Actual behavior**
The following duplicated entries appeared in `.link_files.txt` 
```
C:\my-mbed-project\projects\test-json11\build\debug\.\json11.o
C:\my-mbed-project\projects\test-json11\build\debug\.\mbed_memory_status.o
C:\my-mbed-project\projects\test-json11\build\debug\.\test_env.o
[...]
C:\my-mbed-project\projects\test-json11\build\debug\json11.o
C:\my-mbed-project\projects\test-json11\build\debug\mbed_memory_status.o
C:\my-mbed-project\projects\test-json11\build\debug\test_env.o
```

**Steps to reproduce**
I invoked make.py directly using the following command (on MINGW64 / windows7):
```
[mbed] Exec "C:\Python27\python.exe -u C:\my-mbed-project\libs\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F767ZI --profile C:\my-mbed-project\profiles\debug.json --source C:\my-mbed-project\projects\test-json11 --source C:\my-mbed-project\libs\mbed-memory-status --source C:\my-mbed-project\libs\json11 --source C:\my-mbed-project\libs\mbed-os --build C:\my-mbed-project\projects\test-json11\build\debug --app-config C:\my-mbed-project\projects\test-json11\mbed_app.json" in C:\my-mbed-project
```

Note that after a clean everything works fine, this issue only appears when then doing a second build.

I fixed this by adding
`path = abspath(path)` to the `scan_resources` function in `mbed-os\tools\toolchains\__init__.py`. Because the paths are now normalized the duplicated get filtered by `r.objects = sorted(set(r.objects))` in function `link_program`

Is this a bug or am I doing something horribly wrong?

jorisa, 2017-12-21:
I think I narrowed it down to the following function:
https://github.com/ARMmbed/mbed-os/blob/9ebefcd63d7eefb4cf33ab47991cf4c36e54c553/tools/toolchains/__init__.py#L819

When adding some instrumentation:
```
# THIS METHOD IS BEING OVERRIDDEN BY THE MBED ONLINE BUILD SYSTEM
# ANY CHANGE OF PARAMETERS OR RETURN VALUES WILL BREAK COMPATIBILITY
def relative_object_path(self, build_path, base_dir, source):
    source_dir, name, _ = split_path(source)

    obj_dir = join(build_path, relpath(source_dir, base_dir))
    if obj_dir is not self.prev_dir:
        self.prev_dir = obj_dir
        mkdir(obj_dir)

    print "relative_object_path:"
    print "  relpath     = " + relpath(source_dir, base_dir)
    print "  build_path  = " + build_path
    print "  base_dir    = " + base_dir
    print "  source      = " + source
    print "  obj_dir     = " + obj_dir
    print "  name        = " + name
    print "  result      = " + join(obj_dir, name + '.o')

    return join(obj_dir, name + '.o')
```
The result is:
```
#relative_object_path:
#  relpath     = .
#  build_path  = projects\my-project\build\debug
#  base_dir    = C:\my-repo\libs\mbed-os\features/unsupported\tests\mbed\env
#  source      = C:\my-repo\libs\mbed-os\features/unsupported\tests\mbed\env\test_env.cpp
#  obj_dir     = projects\my-project\build\debug\.
#  name        = test_env
#  result      = projects\my-project\build\debug\.\test_env.o
```

A potential fix is wraping the join with another relpath:
```
obj_dir = relpath(join(build_path, relpath(source_dir, base_dir)))
```

cmonr, 2017-12-28:
Merged.

closed 2017-12-28 by cmonr