ARMmbed/mbed-os #5683: Initialized PwmOut on NRF52 starts writing data immediately

jrobeson, 2017-12-11:
### Description
- Type: Bug
---------------------------------------------------------------
## Bug

**Target**
RBLAB_BLENANO2

**Toolchain:**
GCC_ARM

**Toolchain version:**
gcc 6.3.1
**mbed-cli version:** 
1.2.0

**mbed-os sha:**
182bbd51b

**DAPLink version:**

**Expected behavior**
I expect PWMOut not to write anything to the PWM pin when initialize
**Actual behavior**
It writes a lot that comes out as clicks in the buzzer.
**Steps to reproduce**
Add `PwmOut buzzer(somePin)` to mbed-os-example-blinky.

screamerbg, 2017-12-20:
@cmonr this board uses the nRF52832 implementation. Could you please check if this is a problem on the nRF52-DK? If that's the case then this is a Nordic MCU HAL implementation problem. Otherwise RedBear specific. Please let me know.

cmonr, 2017-12-20:
@screamerbg Finished reproducing the issue on an NRF52-DK. The issue is with the Nordic HAL.

jrobeson, 2017-12-20:
what does HAL mean in this case? the files in `target/`? or `drivers_nrf/hal` ? or something else?

0xc0170, 2018-01-08:
> what does HAL mean in this case? the files in target/? or drivers_nrf/hal ? or something else?

@cmonr Can you please specify ? We should notify proper people about this bug so it could be fixed

cmonr, 2018-01-08:
@jrobeson @0xc0170 The problem is in Nordic's implementation of the PWM Init function, specifically https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_NRF52_COMMON/pwmout_api.c#L151

0xc0170, 2018-01-09:
Thanks @cmonr 

@ARMmbed/team-nordic Please review

nvlsianpu, 2018-01-09:
@0xc0170 - what is expectation to the duty cycle once pwm instance is created? Should it be 0% or 100%.

0xc0170, 2018-01-09:
> @0xc0170 - what is expectation to the duty cycle once pwm instance is created? Should it be 0% or 100%.

The most targets have this as default in the init

```
    // default to 20ms: standard for servos, and fine for e.g. brightness control
    pwmout_period_ms(obj, 20);
    pwmout_write    (obj, 0);
```

@bulislaw FYI, it should be covered in the new specs

bulislaw, 2018-01-09:
Made a note in the PREQ.

marcuschangarm, 2018-03-08:
We have a PR for a new PWM implementation for our feature branch: https://github.com/ARMmbed/mbed-os/pull/6297

It should stay inactive on init until explicitly activated by the user. 

Feature branch progress can be tracked here: https://os.mbed.com/forum/upcoming-features/topic/29477/

0xc0170, 2018-05-18:
Should be resolved on master now after updating drivers. If that is not the case, please reopen with an update

closed 2018-05-18 by 0xc0170