ARMmbed/mbed-os #5686: malloc() / dynamic memory corruption bug

bmcdonnell-ionx, 2017-12-11:
Am I doing something dumb here, or is this a real issue?

### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
LPC4088

Specifically, on an Embedded Artists' [LPC4088 QuickStart Board](https://www.embeddedartists.com/products/boards/lpc4088_qsb.php).

**Toolchain:**
All?
Observed on GCC_ARM and ARM.

**Toolchain version:**
mbed online compiler, used on 2017-12-11
NXP MCUXpresso IDE v10.1.0 [Build 589] [2017-11-14]

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
5f6572179 (HEAD, tag: mbed_lib_rev157, tag: mbed-os-5.6.6, origin/mbed-os-5.6, mbed-os-5.6) Merge pull request #5634 from ARMmbed/release-candidate

**Description**

The code below dynamically allocates an array of ints, writes the index to each location, and reads it back. A portion of the memory gets corrupted, as described in comments.

```
#include "mbed.h"
#include <cstdio>

#define NUM_INTS         (100)
#define FIELD_WIDTH_STR  "3"
#define SERIAL_BAUD_RATE (115200)

Serial s(USBTX, USBRX, SERIAL_BAUD_RATE);

// Demonstrate an apparent bug with malloc() - or am I doing something wrong?
// Memory corruption occurs ~1/4 of the way into the allocated space, and extends for 10 items.
// Vary NUM_INTS to see this pattern.
int main()
{
  //static int nums[NUM_INTS];
  int *nums = (int*)malloc(NUM_INTS);

  printf("\r\n"
         "---\r\n"
         "malloc() Issue Demo\r\n"
         "Using EA LPC4088QSB\r\n" // Embedded Artists' LPC4088 QuickStart Board
         "Build Date: " __DATE__ " at " __TIME__ "\r\n"
         "\r\n");

  // show addresses of allocated memory bounds
  printf("&nums[%" FIELD_WIDTH_STR "s] == %p\r\n", "0", nums);
  printf("&nums[%" FIELD_WIDTH_STR "u] == %p  (one past the end)\r\n",
          NUM_INTS, &nums[NUM_INTS]);

  printf("Load dummy data...");
  fflush(stdout);
  for (size_t i = 0; i < NUM_INTS; i++)
  {
       nums[i] = i;
  }
  printf("done.\r\n\r\n");

  /*
  printf("Dump memory ints to screen:\r\n");
  for (size_t i = 0; i < NUM_INTS; i++)
  {
      printf("%" FIELD_WIDTH_STR "d\r\n", nums[i]);
  }
  */

  bool error = false;

  printf("Check memory...");
  fflush(stdout);
  for (size_t i = 0; i < NUM_INTS; i++)
  {
      if (nums[i] != (int)i)
      {
          error = true;
          printf("\r\nERROR: nums[%" FIELD_WIDTH_STR "u] == "
                  "%" FIELD_WIDTH_STR "d", i, nums[i]);
      }
  }

  if (!error)
  {
      printf("OK.");
  }
  printf("\r\n");

  while (true) {}
}
```

Output:
```
---
malloc() Issue Demo
Using EA LPC4088QSB
Build Date: Dec 11 2017 at 17:10:56

&nums[  0] == 10002050
&nums[100] == 100021e0  (one past the end)
Load dummy data...done.

Check memory...
ERROR: nums[ 26] == 1667590211
ERROR: nums[ 27] == 1847605842
ERROR: nums[ 28] == 1534291317
ERROR: nums[ 29] == 1563963936
ERROR: nums[ 30] == 540884256
ERROR: nums[ 31] == 942683189
ERROR: nums[ 32] == 942748472
ERROR: nums[ 33] == 168430898
```

gte1, 2017-12-11:
It seems like you are allocating 100 bytes.
Change to:
int *nums = (int*)malloc(NUM_INTS*sizeof(int));

bmcdonnell-ionx, 2017-12-11:
Thanks for that. Looks like it was my dumb mistake - verifying now...

bmcdonnell-ionx, 2017-12-11:
Yup, my mistake.

Thanks for your help.

closed 2017-12-11 by bmcdonnell-ionx

cmonr, 2017-12-11:
Hi @bmcdonnell-ionx,

I would suggest replacing `int *nums = (int*)malloc(NUM_INTS);` with `int *nums = (int*)malloc(NUM_INTS * sizeof(int));` and seeing if you still get issues.