ARMmbed/mbed-os #5690: Bug: NRF52 - SPI Speed Fixed to Maximumum (8MHz)

drahnr, 2017-12-12:
### Description
- Type: Bug
- Priority: Major

## Bug
The speed of SPI is fixed to 8MHz despite setting it to less:

```
	csPin = 1;
	spiPins.format(8,0);
        spiPins.frequency(SPI_FREQUENCY_FREQUENCY_M1);
        spiPins.set_default_write_value(0x00);

    irqPin.mode(PullNone);
    irqPin.fall(isr_callback);
	csPin = 0;
```

**Target**
K64F|??

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (Fedora 7.1.0-5.fc27) 7.1.0

**mbed-cli version:** 
(`mbed --version`)
1.2.2

**mbed-os sha:**
(`git log -n1 --oneline`)
6b093eb12 (HEAD -> mbed-os-5.6) Use C++0x
5f6572179 (tag: mbed-os-5.6.6, origin/mbed-os-5.6) Merge pull request #5634 from ARMmbed/release-candidate

**DAPLink version:**

**Expected behavior**
Use the set frequency

**Actual behavior**
Alaywas sticks to 8MHz (Oscilloscope)

**Steps to reproduce**
Set the speed and measure with oscilloscope

0xc0170, 2017-12-13:
@ARMmbed/team-nordic

nvlsianpu, 2017-12-14:
Hi @drahnr, @0xc0170 
SPI::frequency() API is expected to take frequency in [Hz] unit. In snniped of code you shown  i see nordic specific bit-field value SPI_FREQUENCY_FREQUENCY_M1 = 0x10000000UL, so much above 8 MHz (is saturated by nordic driver here: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/spi_api.c#L426).

drahnr, 2017-12-14:
You are right, I miscounted the 0s in that define.

nvlsianpu, 2018-01-02:
@drahnr Did you verified this?

drahnr, 2018-01-03:
Not yet. I hope I can do it this week @nvlsianpu

nvlsianpu, 2018-01-08:
0xc0170 This can be closed.

nvlsianpu, 2018-01-08:
@0xc0170 This can be closed.

closed 2018-01-08 by 0xc0170

drahnr, 2018-01-08:
It works as expected, sorry for the noise.