ARMmbed/mbed-os #5692: How to debug with bootloader?

amq, 2017-12-12:
## Question

I'm happily debugging with `gnuarmeclipse`.

However, after adding `"target.bootloader_img"` to `mbed_app.json` debugging stops working. Looks like that's because the binary is padded, which is the expected behavior for mbed-cli builds, but I'm not sure it makes sense for IDEs. Actually, it won't even let me to export without an existing bootloader binary.

Is there a way to ignore the bootloader for Eclipse?

Or, even better, is there a way to debug with bootloader?

amq, 2017-12-15:
I think something like this could be enough:

```
mbed export -i gnuarmeclipse --no-bootloader -m UBLOX_C030_N211
```

0xc0170, 2017-12-15:
@theotherjimmy @c1728p9 please review

amq, 2017-12-16:
Another workaround:

1. Open the `Debug Configuration`

2. In `Startup` tab, uncheck `Pre-run/Restart reset` and `Set breakpoint at`

3. Set a breakpoint manually in code

This is simple enough to be considered a solution, we would just need to document it.

It also seems to apply for all debug options: Segger, OpenOCD, PyOCD and QEMU.

amq, 2018-01-04:
My last comment worked for C030, but not for EFM32PG12, not sure why.

For EFM32PG12 I had to flash the bootloader first, then debugging in Eclipse works as usual, no need to modify the configuration.

AnotherButler, 2018-01-25:
@theotherjimmy

theotherjimmy, 2018-02-13:
@amq 

>Is there a way to ignore the bootloader for Eclipse?

Don't set `target.bootloader_img`? I think I'm misunderstanding something here.

>Or, even better, is there a way to debug with bootloader?

As eclipse (or any exporter for that matter) does not understand Managed Boot Loader building, you will have to flash the boot loader first, then flash from eclipse. That way there is a boot loader to boot from. I think that should work for all platforms.

amq, 2018-02-13:
@theotherjimmy but how do I build a padded binary without `target.bootloader_img`? Or do you mean removing it temporarily before exporting?

My main point was that we could add an option to ignore bootloader or disable padding when exporting.

> you will have to flash the boot loader first, then flash from eclipse
Right, that's how I'm doing it now, works. At the very least we could mention this in the bootloader documentation.

theotherjimmy, 2018-02-13:
@amq 
>My main point was that we could add an option to ignore bootloader or disable padding when exporting.

Without the boot loader or padding, there is no longer a difference between a boot-loader-using project and a not-boot-loader-using project. That's why I suggested removing `target.bootloader_img`.

Further, I don't want to add such a narrow option. I recall @geky suggesting that we have an option for overwriting Config options on the command line. That feature would implement the same behavior (admittedly, the invocation would be more complex).

> At the very least we could mention this in the bootloader documentation.

👍 I'll make that change.

theotherjimmy, 2018-02-13:
@amq Here's a draft of the text. Does this match your experiences?

### Exporter limitations                                                                                 
                                                                                                         
While the exporters can export a project using the configuration parameters above, there are some limitations.                                                                                                   
                                                                                                         
The exporters do not interpret Mbed OS Configuration, and any changes to configuration parameters, especially boot loader parameter, require you to re-run the `mbed export` command.                            
                                                                                                         
Further, The exporters do not implement the post build merge used in managed boot loader builds. After exporting a project with the `target.bootloader_img` setting, you are responsible for flashing the binary mentioned in the configuration parameter. Without flashing this boot loader image, the device will not boot correctly, and debugging may not work.

theotherjimmy, 2018-02-13:
I included that change with my recent edits to the Boot Loader Docs: https://github.com/ARMmbed/Handbook/pull/408

closed 2018-03-09 by amq