ARMmbed/mbed-os #5699: EthernetInterface fails to connect from other than default thread

amq, 2017-12-13:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
UBLOX_C030_N211

**Toolchain:**
GCC_ARM

**Toolchain version:**
6-2017-q2-update

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
5f6572179d66ce4c09d6517b659ac51133cc980d

**Expected behavior**

Both examples should succeed to connect:

```
#include "mbed.h"
#include "EthernetInterface.h"

int main() {
    EthernetInterface net;
    int result = net.connect();
    while (true);
}
```

```
#include "mbed.h"
#include "EthernetInterface.h"

void test() {
    EthernetInterface net;
    int result = net.connect();
}

int main() {
    Thread *testThread = new Thread(osPriorityNormal, OS_STACK_SIZE * 8, NULL, "test_thread");
    testThread->start(callback(test));
    while (true);
}
```

**Actual behavior**

The second example consistently fails.

0xc0170, 2017-12-13:
@ARMmbed/team-ublox please review

RobMeades, 2017-12-13:
Confirmed that I can reproduce that here, will investigate...

RobMeades, 2017-12-13:
The same thing happens on the generic ST NUCLEO_F429ZI target, so I think this is somewhere in the STM32F4 platform code.  Is this the correct team alias @ARMmbed/team-st-mcd for ST?

jorisa, 2017-12-14:
Same here (NUCLEO-F767ZI):
https://os.mbed.com/forum/bugs-suggestions/topic/28418/

RobMeades, 2017-12-14:
Just to complete the picture, both methods _do_ work on a K64F board, so this isn't a generic Mbed issue it is an ST Mbed platform issue.  Adding some ST people in case the above team alias is not correct: @LMESTM, @jeromecoutant , @bcostm, @adustm.

jsln, 2018-02-01:
I cannot reproduce the issue on a NUCLEO_F429ZI board. I have been trying with release 5.7.3 as well as the commit reported in the bug (sha 5f65721). I am using:

Target
NUCLEO_F429ZI

Toolchain:
GCC_ARM

Toolchain version:
7-2017-q4-major

mbed-cli version:
1.3.0

Please let me know if I can help with some additional test.

bcostm, 2018-02-01:
Thanks @jsln for your help. We are going to make some trials with the 2nd example on different boards and we'll let you know.

0xc0170, 2018-02-01:
> 7-2017-q4-major

might not be relevant, but important - supported version is GCC ARM 6

bcostm, 2018-02-01:
No issue on our side and we use GCC ARM 6.

closed 2018-06-12 by 0xc0170