ARMmbed/mbed-os #5700: Sync with CMSIS repo / Allow use of CMSIS_DSP

bmcdonnell-ionx, 2017-12-13:
### Description
- Type: Enhancement
- Priority: Major | Minor (somewhere in this range)

---------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

There is no straightforward or maintainable way to utilize the parts of CMSIS which are not already included by `mbed-os`, e.g. CMSIS-DSP.

This is due to the subset of files chosen from the CMSIS repo, which are taken from different revisions of the CMSIS repo, organized in a different folder structure, and further customized within `mbed-os`. (In my working copy of `mbed-os`, I did `git log --all --graph --decorate cmsis` to investigate this. I'm not aware of those customizations being pushed upstream.)

**Suggested enhancement**

* Most importantly, *keep in sync with the [CMSIS git repo](https://github.com/ARM-software/CMSIS_5)*. If necessary (due to organizational reasons or whatever), fork it, and maintain your own `mbed` branch.
* Do not rearrange the file/folder structure of CMSIS in the `mbed-os` tree. Optionally remove some files/folders in the mbed-os CMSIS folder subtree, if necessary.
    * Or maybe even somehow pull all of CMSIS in as a sublibrary. Optionally utilize the pre-built library files provided by CMSIS. (This may be a separate later improvement.)

**Pros**

* Easy to integrate bugfixes and enhancements from upstream CMSIS.
* Easy to use excluded parts of CMSIS; or the whole thing will be available.

**Cons**

* Effort / up-front work to implement this change.

bmcdonnell-ionx, 2017-12-14:
For the record, I noticed that CMSIS-DSP appears to be located under `mbed-os/features/unsupported/dsp`. How can I include this in my build?

0xc0170, 2018-05-23:
@ARMmbed/mbed-os-core Please review

SenRamakri, 2018-05-23:
@deepikabhavnani @c1728p9 @bulislaw @kegilbert - Adding team for review.

deepikabhavnani, 2018-05-23:
@bmcdonnell-ionx - Appreciate your inputs on enhancements. 
For now, if you want to know all the changes in Mbed OS from CMSIS please refer to https://github.com/ARMmbed/mbed-os/blob/master/tools/importer/cmsis_importer.json#L2. 
We try to keep the json script up to date, it captures all the files picked from CMSIS repo, and SHA of change commits applied on top of CMSIS.

geky, 2018-05-23:
@mbartling, weren't you looking into using the DSP module with mbed-os at one point?

SenRamakri, 2018-12-28:
@bmcdonnell-ionx - Thanks for raising this issue and we have copied this issue into our internal project tracker. Unfortunately, we don't have plans to immediately address this issue. At this point, I would like to close this out as we may address this in only in future as to how to support CMSIS-DSP library. Please note that if multiple developers start seeing this is a blocking issue we can raise the priority to address this sooner as well. Let us know if you have more feedback on this,

closed 2018-12-28 by SenRamakri

bmcdonnell-ionx, 2019-01-02:
@SenRamakri, since you do still plan to address it (right?), please consider leaving this issue open for visibility purposes.

bmcdonnell-ionx, 2019-01-03:
@SenRamakri, bummer, but understood. Thanks.