ARMmbed/mbed-os #5702: Expose filesystem stats

bmcdonnell-ionx, 2017-12-13:
Reference: [Docs / mbed OS API References / Storage APIs / File system](https://docs.mbed.com/docs/mbed-os-api-reference/en/latest/APIs/storage/filesystem/)

### Description
- Type: Enhancement
- Priority: Minor

---------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

There is no way to retrieve file system statistics (e.g. free space) via the POSIX API, nor via the FileSystem objects. AFAICT, the only way to do it now is to reach all the way down into [filesystem/fat/ChaN/ff.{h,cpp}](https://github.com/ARMmbed/mbed-os/tree/master/features/filesystem/fat/ChaN), and call `f_stat()`.

**Suggested enhancement**

Expose file system stats. This could be done in stages:

1. Add a stat function(s) to `FileSystem.{h,cpp}` and `FATFileSystem.{h,cpp}`.
2. Expose the stats via POSIX API functions [`fstatvfs()` and `statvfs()`](http://pubs.opengroup.org/onlinepubs/9699919799/functions/fstatvfs.html).

Even [1] without [2] is helpful, e.g. if [2] is missing from whatever other library the filesystem feature depends on.

**Pros**

Consistent, simpler API.

**Cons**

??

0xc0170, 2017-12-18:
@geky Please review

geky, 2018-03-14:
I think this was resolved in #5768 

Feel free to reopen otherwise

bmcdonnell-ionx, 2018-03-26:
@geky

> I think this was resolved in #5768
>
> Feel free to reopen otherwise

`statvfs()` was added, but `fstatvfs()` was not. Consider also adding `fstatvfs()` for feature completeness / flexibility. (Minor issue.)

Note: I cannot re-open this issue myself.

SeppoTakalo, 2019-08-26:
I'm closing the internal ticket, we are not going to work on this in the given time.
It can be considered as a future feature requests.

closed 2019-08-26 by ciarmcom