ARMmbed/mbed-os #5705: Question regarding priorities of System Timer Events and Send/Receive Events

ghseb, 2017-12-14:
### Description
- Type: Possible Bug
- Priority: Major

---------------------------------------------------------------
## Bug

When a timer event is created via `eventOS_event_timer_request()` the priority `ARM_LIB_MED_PRIORITY_EVENT` is assigned in system_timer.c. Send/Receive Events get the priority `ARM_LIB_HIGH_PRIORITY_EVENT` in `send_receive_event()`.
When there is high load an a network connection (Cellular with blockwise transfers in our case) such many Send/Receive Events are generated that Timer Events (which Registration Updates depend on) are not emitted and the Registration times out.
Our question is if this is expected behaviour and why Send/Receive Events should not have the same priority than Timer Events? 

In our case this solved the issue, but there may be drawbacks we did not encounter yet.

Regards,
Sebastian

Roger-Strassburg, 2017-12-14:
A note on the scenario described here. While transferring a 400k-byte file from the server to the client, registration updates were being queued on the event queue, but were getting starved out by the receive events. This prevented registration updates from getting sent before timing out, which in turn invalidated the file transfer.

Changing the priority of timer events in system_timer.c to be the same as receive and transmit events fixed this.

0xc0170, 2017-12-18:
@kjbracey-arm please review

kjbracey-arm, 2017-12-18:
Note you do have the flexibility to request an event at the priority of your choice using `eventOS_event_timer_request_in`/`eventOS_event_send_in` or the similar `after`/`at`/`every` API.

I suspect you're right though, something is using high priority that probably shouldn't. I think a bit of priority creep has happened. I'd suggest the person using HIGH should try lowering it. (Is that the mbed client code?)

Mind you, if it's a CoAP blockwise transfer over cellular, I'm surprised there's no idle time such that the event queue empties to let your timer trigger. Surely after sending a block request, you would be idle waiting for a round-trip time?

Is there possibly some other bug here such that the event queue is getting jammed or continuously refilled by mistake?

In general, I've found the event queue priorities to be not that useful, but in general also quite harmless, as the event queue is not normally that constantly busy, even when streaming data.

ghseb, 2017-12-20:
Thank you for your reply.
You are right. It is regarding mbed-client code and therefore we would like to avoid local modifications. According to your suggestion, we reduced the amount of Receive Events that are triggered. This seems to help. Also, enabling duplication detection seems to reduce the load of the event queue. Currently we do not observe the described issues anymore.
Still, we belief that Send/Receive Events should have lower priority than events that are by definition bound to a specific instance of time.

We dont know implications to other use cases, so do you recommend to file a separate bug report?

Regards,
Sebastian

kjbracey-arm, 2017-12-20:
A couple of layers of the code do have "repeat event" suppression to try to avoid jamming up the slower parts of the system with lots of unnecessary events - this occurs in mbed OS Socket for generating application sigios from stack callbacks, and I thought it also occurred in mbed client's connection handler.

Every time you're downgrading to a lower priority, eg from interrupt to thread, it's worth doing this. Can avoid overflowing queues - (eg https://github.com/ARM-software/CMSIS_5/issues/283)

You could raise this issue with the mbed client guys to suggest the priority lowering - I agree MEDIUM seems more appropriate for their receive event (or indeed, almost any event...). I don't know if they made it HIGH to overcome some problem.

cmonr, 2018-07-26:
Closing since it seems like this question was answered.

closed 2018-07-26 by cmonr