ARMmbed/mbed-os #5706: Mbed TLS example benchmark build fail for IAR + ODIN

mazimkhan, 2017-12-14:
mbed-os-example-tls/benchmark fails to build for IAR + ODIN

### Description
- Type: Bug (build failure)
- Related issue: https://github.com/ARMmbed/mbed-os-example-tls/issues/133#issuecomment-351697974
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
IAR

**Toolchain version:**
7.8 

**Steps to reproduce**
See issue https://github.com/ARMmbed/mbed-os-example-tls/issues/133#issuecomment-351697974

## Question

**How to?**
See issue https://github.com/ARMmbed/mbed-os-example-tls/issues/133#issuecomment-351697974

0xc0170, 2017-12-15:
@ARMmbed/team-ublox Please review

andreaslarssonublox, 2018-01-09:
@0xc0170 See comment in https://github.com/ARMmbed/mbed-os-example-tls/issues/133

RonEld, 2018-09-12:
I couldn't reproduce this. Perhaps it's an IAR version issue?

screamerbg, 2018-09-19:
@RonEld Thanks for investigating. Closing this issue. Feel free to reopen if the issue persists with Mbed OS 5.9.4 or later releases.

closed 2018-09-19 by screamerbg