ARMmbed/mbed-os #5709: Nucleo-L432KC Drag-and-Drop Programming of .bin files not always working

jvn314, 2017-12-14:
### Description
- Type: Bug 
- Related issue: https://os.mbed.com/forum/bugs-suggestions/topic/28258/?page=1#comment-53688
- Priority: Blocker 

## Bug
**Target**
STM Nucleo-L432KC

**Toolchain:**
Mbed Online Compiler

**Toolchain version:**
mbed 2.0 Library Revisions 153,154,155,156,157 (Revision 152 seems OK)

**Operating system:**
Windows 8.1 fully patched

**ST-LINK, ST-LINK/V2, ST-LINK/V2-1 USB driver signed for Windows7, Windows8, Windows10**
USB Driver downloaded on 12/14/2017 from http://www.st.com/en/development-tools/stsw-link009.html
Drivers installed and PC rebooted on 12/14/2017.

**ST-LINK, ST-LINK/V2, ST-LINK/V2-1 firmware upgrade**
Latest firmware downloaded on 12/14/2017 from http://www.st.com/en/development-tools/stsw-link007.html.
Firmware installed using: STLinkUpgrade 3.2.2.
Firmware version: V2J28M18.
Board unplugged after reflash, re-inserted, firmware verified as V2J28M18.
Verified that only ONE Nucleo board is present on system, i.e. correct target is being flashed.

**Expected behavior**
Drag-and-Drop programming of .bin files generated by mbed online compiler to USB Mass Storage device "NODE_L432KC" should always work when correct compile target is being used (Nucleo-L432KC).

**Actual behavior**
Drag-and-drop programming does *not* always work.  Some .bin files are *not* flashed to the target.

**Steps to reproduce**
1. Select Nucleo-L432KC as target, and use online compiler to compile this code: https://os.mbed.com/users/nixnax/code/PPP-Blinky/ 

2. Attempt to save mbed-generated .bin file to USB MSD drive named "NODE_L432KC".
The flash fails. The normal red/green LED flash sequence that occurs during flashing does not occur.

**Additional Information**
IF the first four bytes in the .bin file are changed (via hex editor) to 00 C0 00 20 then the .bin file can be successfully saved to the USB MSD, flashing appears to be successful and the code appears to execute correctly.  So the issue seems to be related to the first 32-bit word in the .bin file.

0xc0170, 2017-12-14:
@ARMmbed/team-st-mcd please review

jeromecoutant, 2017-12-14:
Hi

> Board unplugged after reflash, re-inserted, firmware verified as V2J28M18.

Up to date version is 2.29.18
Before this version NUCLEO_L432KC can't boot

Regards,

jvn314, 2017-12-14:
>  to date version is 2.29.18 
> Before  this version NUCLEO_L432KC can't boot

Thank you! Confirming that latest firmware upgrade from 
http://www.st.com/en/development-tools/stsw-link007.html
works.

Not sure how i got to 2.28.18, but I think it's because all versions are called en.stsw-link007.zip and I grabbed a previous download to unzip.

Thanks for the quick response. All good now.

closed 2017-12-14 by 0xc0170