ARMmbed/mbed-os #5711: mbed can't find a header file in a folder

ismailhkose, 2017-12-15:
- Type: Bug
- Priority: Blocker

I developed a class and header for my device and put it in a folder in the root directory, then included the header file in main,cpp; however, it cant find the header file. I think there might be a bug in header name file parser in mbed-cli/mbed. When I move them to in root folder. It finds and builds successfully.

---------------------------------------------------------------
## Bug
Compile [ 18.3%]: main.cpp
[Fatal Error] main.cpp@3,0: my_header.h: No such file or directory
[ERROR] /path/main.cpp:3:25: fatal error: my_header.h: No such file or directory
compilation terminated.

**Toolchain:**
GCC_ARM

**Toolchain version:**
GCC 4.7.4

**mbed-cli version:** 
1.2.0 - 1.2.2

**mbed-os sha:**
182bbd51bc8d1b6217477c2b5de3f1d0c5e6249f

0xc0170, 2017-12-15:
Can you use verbose for compilation so we can review what paths are there? I would  check if your /path is in the include paths first.

how is your header file included in the code? What include paths are set?

cmonr, 2018-07-26:
Closing since this should now work. 

Please open a new issue if this is not the case.

closed 2018-07-26 by cmonr