ARMmbed/mbed-os #5713: nRF52 does not go to sleep

janjongboom, 2017-12-15:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
nRF52-DK

**Toolchain:**
ARM / GCC_ARM

**mbed-os sha:**
Mbed OS 5.6.6

**Expected behavior**

1.5 uA current in sleep mode, as observed in older versions of Mbed

**Actual behavior**

900 uA current in sleep mode

https://os.mbed.com/questions/79731/nrf52-mbed-os-huge-power-consumption

0xc0170, 2017-12-15:
@pan- @ARMmbed/team-nordic please review

pan-, 2017-12-15:
The code exhibited by the user is incorrect because it is the RTOS determine when the system is able to go to sleep. Calling WFE/WFI directly is begging for trouble. 

Similarly, the function `sleep` should not be used by application code when the RTOS is present. This function put the system to sleep until an interrupt is triggered; the ticker used by the RTOS kicks in every ms. In other word, the system will wake up almost immediately. 

To put the system to sleep when the RTOS is present, use the RTOS primitives (Thread::wait, Thread::signal_wait, Semaphore, Mail). It is dangerous and inefficient to circumvent the system.

Note that sleep behavior exhibited by the NRF52 targets on previous versions of mbed was buggy! Nordic physical implementation of WFE/WFI shutdown the systick clock upon sleep entry; this was not the expected behavior.  On mbed OS 5.6.6, I measure a power consumption of 1.7uA for the following code: 

```c++
int main() {
    while (true) {
        Thread::wait(0xFFFFFFFF);
    }

    return 0;
}
```

closed 2017-12-15 by pan-

JuricaResetar, 2017-12-16:
@pan- Thank you for your reply. I completely understand your answer and thank you for it. But I'm not sure about the code you provided me with. I used it and my power consumption is miles from the desired 1.7uA. And I'm very sure the problem is not in HW (with older version of mbed-os I can get 1.7uA).

pan-, 2017-12-18:
@JuricaResetar It'd have been interesting to mention the compiler and profile used. The mbed OS mentioned exhibit a different power consumption when the *develop* profile is used with the ARM compiler.

|         | GCC   | ARM   | IAR   |
|---------|-------|-------|-------|
| develop | 1.7uA | 480uA | 1.7uA |
| release | 1.7uA | 1.7uA | 1.7uA |
 
I'm not sure this qualify as a bug. Additional processing made by the OS when compiled with the develop or debug profile is not representative of release mode.

JuricaResetar, 2017-12-18:
@pan- I'm not sure what you mean by develop and release profile... I use default mbed online compiler (ARM) and I got around 480uA. Do you know how to switch it to release profile?

pan-, 2017-12-18:
@JuricaResetar The mbed online compiler use the develop profile and the ARM compiler. You can mimic the release profile by adding the following macros (compile -> compile macros): 

```
NDEBUG
__ASSERT_MSG
```

JuricaResetar, 2017-12-18:
@pan- Thank you very much sir.

janjongboom, 2017-12-18:
@pan- Awesome, didn't know we had the compile macros there.

pan-, 2017-12-18:
@janjongboom Thanks for opening a ticket against the online compiler. 

In the meantime having well articulated requirements related to power consumption, code size, semi hosting or sleep behavior in each of the profile mbed-os offer would help a lot to decipher what's a bug. It would be equally important to understand what is the commitment made by the mbed os team to preserve behavior between release and how we ensure it. 

@sg- Any pointers ?

janjongboom, 2017-12-18:
I'm also wondering if this is documented somewhere @iriark01 @AnotherButler

pan-, 2017-12-18:
@janjongboom I've found [this](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/build_profiles/). It is outdated regarding sleep specification (it doesn't mention deep sleep and how it is linked to profiles) but the difference with debug is accurate.

AnotherButler, 2017-12-18:
We have the latest version of the build profiles document: https://os.mbed.com/docs/v5.6/tools/build-profiles.html

We also have the sleep manager documentation: https://os.mbed.com/docs/v5.6/reference/sleep-manager.html

Are these what you're looking for?

janjongboom, 2017-12-18:
@AnotherButler Great, the sleep manager docs are a bit thin, but good that it's documented. It should be more clear that `sleep()` should never be called by hand.

0xc0170, 2017-12-18:
> @AnotherButler Great, the sleep manager docs are a bit thin, but good that it's documented. It should be more clear that sleep() should never be called by hand.

Patches welcome!  Or at least create an issue for these

janjongboom, 2017-12-18:
@0xc0170 See https://github.com/ARMmbed/Handbook/issues/365