ARMmbed/mbed-os #5717: Bug - SPI::abort_transfer() can decrement/unlock deep sleep manager lock without ever incrementing/locking it

scartmell-arm, 2017-12-15:
### Description

- Type: Bug
- Related issue: `#5716`
- Priority:  Minor

When transferring data using `SPI::start_transfer()` deep sleep mode is disabled by calling `sleep_manager_lock_deep_sleep()`. When the transfer is complete or aborted `sleep_manager_unlock_deep_sleep()` is called decrementing the lock count counter, and potentially re-enabling deep sleep mode.

The deep sleep manager tracks the number of locks acquired by device drivers, when `lock_deep_sleep()` is called the lock count is incremented, when `unlock_deep_sleep()` is called the count is decremented, while the `count != 0` deep sleep mode is disabled.

Each device driver should call `lock` and `unlock` as a pair of operations to prevent either decrementing/removing a different device drivers lock or locking deep sleep permanently. `SPI::abort_transfer()` can be called at any time and will unlock deep sleep mode regardless of whether SPI has previously locked it.

The SPI class should track whether or not it currently holds a lock in the deep sleep manager and only call sleep_manager_unlock_deep_sleep() if it does.

---------------------------------------------------------------
**Target**
All

**Toolchain:**
All

**mbed-os sha:**
`4d81eadb25a509d82d54439c5674843779d30470`

0xc0170, 2017-12-18:
+1, can you send a patch @scartmell-arm to fix this ?

PWMout has similar concept that would fix this, `_deep_sleep_locked` boolean

0xc0170, 2018-01-12:
#5721 and #5722 fixes this

closed 2018-01-12 by 0xc0170