ARMmbed/mbed-os #5718: STM32F7 bootloader does not work

tery123, 2017-12-17:
### Description
- Type: Question
- Priority: Minor

---------------------------------------------------------------
## Question
According to https://os.mbed.com/questions/79720/STM32F7-bootloader-help/ 
**STM32F746ZG will entry Infinite_Loop**
I just only know trace a project in a time. I only to know the code will stop in Infinite_Loop.
And I also don't know this issue publish over here is appropriate not?

0xc0170, 2017-12-18:
Just use https://github.com/armmbed/mbed-os-example-bootloader, and regular blinky to update rather than use bootloader-blinky (it should be deprecated, I'll fix that).

@ARMmbed/team-st-mcd Please review

tery123, 2017-12-18:
@0xc0170 It looks like not work, and still entry Infinite_Loop.

bcostm, 2017-12-19:
As of today the bootloader has not been activated on the NUCLEO_F746ZG. This could explain your problem.

tery123, 2017-12-20:
@bcostm I got that and thank you. I will go next phase first.

bcostm, 2017-12-20:
FYI I am actually adding the bootloader on this board (NUCLEO_F746ZG) and testing it with the mbed-os-example-bootloader. Stay tune.

raghidmardini, 2018-01-08:
@bcostm @tery123 I am seeing the same behavior on NUCLEO_F767ZI. Any updates regarding this issue? Thanks!

bcostm, 2018-01-10:
Hi
We're working on it. We should send a fix soon.

jeromecoutant, 2018-01-24:
As #5821 is now merges, I think we can close the issue ?
Thx

bcostm, 2018-01-30:
@0xc0170 can you please reopen it as my PR has been reverted...

bcostm, 2018-02-09:
ST_TO_BE_CLOSED

closed 2018-02-09 by 0xc0170