ARMmbed/mbed-os #5725: Cannot use `unistd.h`

bmcdonnell-ionx, 2017-12-18:
### Description
- Type: Bug
- Priority: Major | Minor

I'm trying to use `fsync()` and `fdatasync()`, but I can't build when I `#include "mbed.h"` and `#include <unistd.h>` in my source file.

---------------------------------------------------------------
## Bug

**Target**
LPC4088

**Toolchain:**
GCC_ARM

**Toolchain version:**
MCUXpresso IDE v10.1.0 [Build 589] [2017-11-14]

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
5f6572179 (tag: mbed_lib_rev157, tag: mbed-os-5.6.6, origin/mbed-os-5.6) Merge pull request #5634 from ARMmbed/release-candidate

**Expected behavior**

Successful build, and `fsync()`/`fdatasync()` flushing to SD Card when I use [sd-driver](https://github.com/ARMmbed/sd-driver).

**Actual behavior**

Fails to build. Name collision between

`unsigned _EXFUN(sleep, (unsigned int __seconds ));` in `unistd.h` and
`__INLINE static void sleep(void)` in `mbed_sleep.h`.

```
Invoking: MCU C++ Compiler
arm-none-eabi-c++ -std=gnu++98 -DMBED_BUILD_TIMESTAMP=1513190426.2 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_DEBUG -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DTARGET_LPC408X -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_MCU_LPC4088 -D__FPU_PRESENT=1 -DDEVICE_PORTIN=1 -DTARGET_LPC4088 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DARM_MATH_CM4 -D__NEWLIB__ [snip]
In file included from c:\nxp\mcuxpressoide_10.1.0_589\ide\tools\arm-none-eabi\include\string.h:10:0,
                 from C:/dev/test-fat32/mbed-os/rtos/TARGET_CORTEX/rtx5/rtx_lib.h:31,
                 from C:/dev/test-fat32/mbed-os/rtos/TARGET_CORTEX/mbed_rtos_storage.h:43,
                 from C:/dev/test-fat32/mbed-os/rtos/rtos.h:28,
                 from C:/dev/test-fat32/mbed-os/mbed.h:38,
                 from ../SDCardTest.cpp:16:
c:\nxp\mcuxpressoide_10.1.0_589\ide\tools\arm-none-eabi\include\sys\unistd.h:209:10: error: conflicting declaration of C function 'unsigned int sleep(unsigned int)'
 unsigned _EXFUN(sleep, (unsigned int __seconds ));
          ^
In file included from C:/dev/test-fat32/mbed-os/drivers/PwmOut.h:24:0,
                 from C:/dev/test-fat32/mbed-os/mbed.h:80,
                 from ../SDCardTest.cpp:16:
C:/dev/test-fat32/mbed-os/platform/mbed_sleep.h:131:22: note: previous declaration 'void sleep()'
 __INLINE static void sleep(void)
                      ^~~~~
make: *** [SDCardTest.o] Error 1
```

**Steps to reproduce**

In a source file:

```
#include "mbed.h"
#include <unistd.h>
```

pan-, 2017-12-18:
Even if `unistd.h` is  accessible by GCC's newlib; it is not the case for other compilers like IAR or the ARM compiler because `unistd.h` is not a header from the C standard library.

It would be impossible to support all features exposed by `unistd.h` like the `exec*` functions however it may make sense to provide `fsync` and `fdatasync`. @kjbracey-arm @geky What are your thoughts ? 

In the meantime you can use `fflush` that flush buffers out of the libc.

bmcdonnell-ionx, 2017-12-18:
Seems I can't use the functions I was trying to anyway. I tried to wrap them as shown below, but the linker complains of
`undefined reference to 'fsync'` and
`undefined reference to 'fdatasync'`.

`wrap-unistd.h`:
```
namespace workaround
{
   int fsync    (int);
   int fdatasync(int);
}
```

`wrap-unistd.cpp`:
```
#include <unistd.h>

static int wrapped_fsync(int fd)
{
   return fsync(fd);
}

static int wrapped_fdatasync(int fd)
{
   return fdatasync(fd);
}

namespace workaround
{
   int fsync(int fd)
   {
      return wrapped_fsync(fd);
   }

   int fdatasync(int fd)
   {
      return wrapped_fdatasync(fd);
   }
} // namespace workaround
```

bmcdonnell-ionx, 2017-12-18:
I'm currently working-around the issue by closing and re-opening the file. However, it seems that at least in Linux land, even this isn't guaranteed to write to disk.

> ### Notes
> Note that **fclose()** only flushes the user-space buffers provided by the C library. To ensure that the data is physically stored on disk the kernel buffers must be flushed too, for example, with **sync(2)** or **fsync(2)**.

https://linux.die.net/man/3/fclose

geky, 2017-12-18:
> I'm currently working-around the issue by closing and re-opening the file. However, it seems that even this isn't guaranteed to write to disk. (At least in Linux land.)

That's pretty interesting. I guess it makes sense when you consider how much of a performance benefit lazily writing to disk is.

Looking at the open group docs, [`fclose`](http://pubs.opengroup.org/onlinepubs/009695399/functions/fclose.html) does have this note:
> The fclose() function shall perform the equivalent of a close() on the file descriptor that is associated with the stream pointed to by stream.

Which for us, retargets to the FileHandle::close function. At that point it's up to the filesystem. Currently, FATFileSystem does write out to disk, but this is not guaranteed if the implementation changes. For LittleFileSystem, writing out to disk during a close is guaranteed.

**TLDR:**
- FATFileSystem close == sync - not guaranteed
- LittleFileSystem close == sync - guaranteed

---

> But mbed OS exposes / utilizes other POSIX functions in the filesystem arena, like open(), close(), read(), etc. Reference: Docs / mbed OS API References / Storage APIs / File system

The mention of POSIX was a shortsighted mistake on my part. We've since removed it, although the changes are staged and not yet visible. We only guarantee the standard C/C++ file API.

That being said, the standard C/C++ file API is frustrating. It provides some file operations, but as you've noticed it quickly hits walls where functionality is missing. Other examples are the complete lack of directory operations or filesystem stat functions.

To make things messier, the different toolchains we support are each provided different subsets of the POSIX APIs. The examples you mentioned (open, close, read) are all provided by GCC, they won't work if you switch to the IAR or ARM toolchain.

Because of these problems, some of the POSIX APIs were introduced to cover common areas where users were relying on GCC provided functions. Most notably, errno error codes and directory operations. Although, as @pan- points out, some POSIX functions are a bit unreasonable in mbed. So, in the end, we don't have a clear boundary. To avoid issues with backwards compatibility, it's better for mbed to be conservative here.

---

**So how do you sync?**

It looks like with the current API, you can't. This is a problem.

There are two fixes I can think of:

**Option 1:**

Provide `fileno` and `fsync` (and `fdatasync`).

This conforms to the expectations of Linux users, but opens the question of where is `open`, `close`, and all the other functions that use filenos.

**Option 2:**

Provide some sort of `FileHandle *mbed_get_filehandle(FILE *)` function to access the underlying mbed File object. This lets you call `File::sync` directly.

@kjbracey-arm, I remember you were looking into this at some point. Did you find a way to convert `FILE*` -> `FileHandle*`?

IMO, option 2, the `mbed_get_filehandle` is more maintainable in the long run. Though I'm open to feedback.

---

As a workaround, @bmcdonnell-ionx have you considered using the C++ mbed file APIs directly? This would give you access to the low-level `File::sync` function directly:

``` cpp
FATFileSystem fs;
File f;
f.open(&fs, "test.txt", O_RDWR);
f.write("Hello World!", 12);
f.sync();
f.close();
```

bmcdonnell-ionx, 2017-12-19:
@geky 

[littlefs](https://github.com/ARMmbed/mbed-littlefs) - interesting. I hadn't heard of that.

> The mention of POSIX was a shortsighted mistake on my part. We've since removed it, although the changes are staged and not yet visible. We only guarantee the standard C/C++ file API.

Are you referring to the reference I linked to? It's more than a mention there. Could you clarify? Which parts of the POSIX filesystem API do you (plural) intend for mbed OS to provide? If any.

> As a workaround, ...have you considered using the C++ mbed file APIs directly?

I think I'll stick with my current workaround for now, but it's good to have options - thanks.

geky, 2017-12-19:
@bmcdonnell-ionx, [littlefs](https://github.com/ARMmbed/mbed-os/tree/master/features/filesystem/littlefs) is also in the master mbed-os development branch. It should be in mbed-os-5.7 once the release is out.

> Are you referring to the reference I linked to? It's more than a mention there. Could you clarify? Which parts of the POSIX filesystem API do you (plural) intend for mbed OS to provide? If any.

The best I can say right now is that we support the standard C99 and C++03 API (anything from [here](http://en.cppreference.com/w/c/io)). As well as any existing functions declared in [mbed_retarget.h](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_retarget.h) until a major version bump.

We need to put together an official list along with the appropriate documentation, but we don't have that available yet.

bmcdonnell-ionx, 2017-12-19:
@geky, if I understand correctly, you're basically saying that "*'[these](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_retarget.h)'* are the subset of POSIX functions we support" (paraphrased). But I'm asking what is the *intent*?

I've inferred that the intent was to present a complete API through which files, filesystems, directories, etc. can be managed and manipulated. To be as standards-compliant and portable as possible. Thus you support all the [C file I/O functions](http://en.cppreference.com/w/c/io), and add in the POSIX functions where there are gaps (e.g. directory management).

Have I understood correctly? If so, then wouldn't your "option 1" be most consistent with that intent? If not, what am I missing?

kjbracey-arm, 2017-12-19:
@geky - `FileHandle *mbed_get_filehandle(FILE *)`.

That takes two steps - the `FILE` knows an `int`, and we can map an `int` to a `FileHandle` (except for `stdin`/`stdout`/`stderr` now, but those would also be `FileHandle`s after #5571).

In #5571, `fileno` for `FILE`->`int` isn't available - we'd need to hack some of the opaque `FILE` structures to get it :( It does have an internal `FileHandle *get_fhc(int)` there, which includes the auto-init of the console, but I didn't bother exposing it. It would make sense to, I guess, but without `fileno` too it would be of somewhat limited usefulness. I guess turning `int` to `FileHandle` would let you downcast the `FileHandle` to something, and there are a couple of `FileHandle` APIs that #5571 didn't expose as POSIX like `set_blocking`.

kjbracey-arm, 2017-12-19:
@geky 

> The examples you mentioned (open, close, read) are all provided by GCC

Are they really? They're present in the `<unistd.h>` header file in the toolchain, but there's no library implementation, I hope. That allows #5571 to provide them.

Only library collision for #5571 I'm aware of is IAR providing `fdopen` itself - I have to provide it for the other 2. That's a different case though, as that's logically a C library function like `fileno`, versus `open`, which is a system function.

bmcdonnell-ionx, 2017-12-19:
@kjbracey-arm, cc @geky,

> I believe most people haven't seen a pressing need to have the POSIX (`int fd`) file operations as well as the C (`FILE`) ones

The only reason I was asking for it now is to have `fsync()` (and similar) operations.

> Even with that, there's unfortunately no way to `fsync` given only a `FILE`, as not all the C libraries have `fileno`, so you can't portably map from `FILE` to `int` or `FileHandle`. Maybe we could be able to come up with something though - at the minute we're living with the limitations of the off-the-shelf libraries. `fileno` has to be implemented by the C library (not all do), or they have to expose their `FILE` structure (not all do), or we'd have to figure it out by reverse engineering...

None of those sound like maintainable options, IMO.

> after #5571 this POSIX-portable code would work:
>
> ```
> int fd = open("/sd/test.txt", O_RDWR);
> FILE *stream = fdopen(fd, "w");
> fprintf(stream, "Hello World!");
> fflush(stream);
> fsync(fd);
> fclose(stream);
> ```
> (Note you have to fflush before fsyncing to empty the C library buffers first).

That's pretty good!

I have an idea on how you maybe able to get the following more canonical form to work...

```
// no open()
FILE *stream = fopen("/sd/test.txt", "w");
fprintf(stream, "Hello World!");
fflush(stream);
fsync(fileno(stream)); // use fileno()
fclose(stream);
```

For the libraries that implement `fileno()`, use their implementations (via `#ifdefs`.) For those that don't, you could provide the `fileno()` function. Have a map of `FILE*` -> `fd`s inside `FileSystem`, so that when the user calls `fileno()`, you can look it up. (Of course the map would *not* be `std::map`; likely just a pair of arrays.)

Would this do the trick? Is it too much overhead?

If this is doable, I don't know if there would be any value add for also implementing `FileHandle *mbed_get_filehandle(FILE *)`.

kjbracey-arm, 2017-12-20:
> Have a map of FILE* -> fds inside FileSystem

The problem is the C library never tells the retarget code its `FILE`. `fopen` calls our `_sys_open`, giving us a filename. We translate the filename to a `FileHandle` and `int`, and we give the C library back the int. It must store the `int` in its `FILE`, but we don't get to find out what `FILE` our `int` got put in!

One pretty sick trick would be for `fileno(FILE*)` to make a dummy `ftell(FILE*)` call or something, and find out which integer the C library then passes to our `_sys_seek`. Could be tricky to synchronise, and the call could conceivably have side-effects or be refused for some reason.

Back to the motivation for the `fsync` - I would note that in most normal applications explicit `fsync` probably shouldn't be needed. Synchronisation being a good idea is more often a property of the storage medium or the system, rather than a specific file. So I would expect the system to have some policy to automatically ensure flushes to disk at appropriate intervals for normal apps. If people are feeling the need to put `fsync` in too much, there might be some sort of general policy problem.

I'd suggest it should normally only be explicitly needed in unusual cases like when you are keeping a file open long-term, but making occasional periodic updates.

bmcdonnell-ionx, 2017-12-20:
> Back to the motivation for the `fsync` - I would note that in most normal applications explicit `fsync` probably shouldn't be needed. Synchronisation being a good idea is more often a property of [...]

On mbed OS 5.6.6, I have a program that was keeping a log file open. It does an `fprintf()` to the log file every few to several seconds, and calls `fflush()` after most of them. The file was created, but *nothing* was ever written to it. (Whereas, I was trying to keep it in sync with other files that were being written.)

Is this a "policy problem" with mbed OS? Has it changed in 5.7.0?

> The problem is the C library never tells the retarget code its `FILE`. `fopen` calls our `_sys_open`, giving us a filename. We translate the filename to a `FileHandle` and `int`, and we give the C [...]

Yeah, that does sound tricky.

Could you always count on `ftell()` calling your `_sys_seek()` in every case? Are there any other lighter-weight functions that call it?

kjbracey-arm, 2017-12-21:
> The file was created, but nothing was ever written to it.

Really? I'm sure it must have been writing the data, if you'd written any moderate amount - there's not unlimited buffering. Maybe it just never updated the directory, so the file was still recorded as zero length? Perhaps a disk check on the media would reveal your missing untracked data.

I'd say a log file is very much a tricky case - obviously you want to see stuff if the system dies without you closing it (and maybe you would never naturally close it, if you assume you will run forever or crash), but on the other hand lots of excess syncing will (I assume) cost you wear on your media.

I'm not familiar with the filesystem implementations - that's @geky's area. My hand wavy suggestion (easy to say as someone who doesn't have to implement it) is that a filesystem should have some sort of configurable ability to automatically periodically sync, in case of crash or sudden media removal. Directory updates reflecting file extension shouldn't be deferred indefinitely unless specifically asked to prioritise speed/wear - probably a good idea to have the directory always indicate the entire map space allocated for the file while open, possibly overestimating the real file content.

Manually closing and reopening your log periodically should help encourage directory updates and maybe data flushing. Whether that is a lighter or heavier performance/wear option than an fsync will depend on the particular filing system.

> Could you always count on ftell() calling your _sys_seek() in every case? Are there any other lighter-weight functions that call it?

Not 100% sure, and not thought of anything yet :/ 

My current view is that as we have a limited number of toolchains to support, we could go for the "know the layout of the FILE and peek" approach, and make very sure we have a CI test that would alert if that ever failed.

bmcdonnell-ionx, 2017-12-21:
> Really? I'm sure it must have been writing the data, if you'd written any moderate amount - there's not unlimited buffering. Maybe it just never updated the directory, so the file was still recorded [...]

I just retried without the workaround, then did `chkdsk /f` on Windows 10. You are correct, the data is there, just not reflected by the metadata - but then with some junk appended to the file, due to the recovery process.

Is the metadata inconsistency a bug, or expected behavior?

> My current view is that as we have a limited number of toolchains to support, we could go for the "know the layout of the FILE and peek" approach, and make very sure we have a CI test that would alert if that ever failed.

If I understand correctly, here you're talking about retrieving `fd`s from `FILE *`s to allow `fsync()`. I know I'm the one that raised the issue, but I'd have to ask if the risk is worth the payoff with your proposed solution. Proceed with caution on that, IMO.

Maybe your previously-proposed workaround isn't so bad for someone who needs lock-step consistency with other files. Is there a target mbed OS version # for having #5571 merged?

kjbracey-arm, 2018-01-02:
> Is the metadata inconsistency a bug, or expected behavior?

It arises from naturally from the implementation, and I think it's working as intended - it must write data relatively often due to lack of buffering, and the effort isn't wasted if it's a continuous data stream - each sector is written exactly once. But repeatedly updating the directory ("metadata") to show current file length is potentially wasted runtime, and wasted storage life.

The current implementation is a reasonable one, and the simplest - maximise storage life by not updating the directory info until file close or sync. If we never get around to closing, chkdsk can figure it out afterwards.

I would say you could make a feature request to ask for a different (configurable?) policy that tries to make it a little more crash-friendly, at the expense of media life.

I would be inclined to have a simple rule - every time more disk space is allocated for the file (ie the map is modified), also modify the directory to reflect that. So number of directory updates = number of map updates. The total media writes could be kept the same by doubling(?) the amount of map space claimed on each file extension.

After crash, that would produce the same result you get now from chkdsk - you will have the data written (aside from stuff left in C and FileSystem buffers) plus trailing junk representing the reserved-but-not-yet-used map space. The map and directory will always be consistent, so chkdsk shouldn't report errors. `close` will write the exact file size to the directory, and that will never be an increase.

I'm a bit fuzzy on how such a scheme should interact with `fsync`. Could be rough if `fsync` wrote current exact file size, and next `write` rewrote to current map size. But then, `fsync` is allowed to be expensive.

(Proviso - I'm not familiar with the FAT filesystem, so might be wrong about the relative costs of map and directory operations.)

> Is there a target mbed OS version # for having #5571 merged?

Not currently. If you want to express interest there, it can't hurt.

bmcdonnell-ionx, 2018-01-02:
> I would say you could make a feature request to ask for a different (configurable?) policy that tries to make it a little more crash-friendly, at the expense of media life.

I think that `fflush()` with `fsync()` / `fdatasync()` will be sufficient.

bmcdonnell-ionx, 2018-03-26:
> after #5571 this POSIX-portable code would work:
>
> ```
> int fd = open("/sd/test.txt", O_RDWR);
> FILE *stream = fdopen(fd, "w");
> fprintf(stream, "Hello World!");
> fflush(stream);
> fsync(fd);
> fclose(stream);
> ```
> (Note you have to fflush before fsyncing to empty the C library buffers first).

#5571 has been merged into mbed-os-5.8, and I'm using this method now. Thanks!