ARMmbed/mbed-os #5730: [NUCLEO-L476] 921600 baud does not work

andrewc-arm, 2017-12-19:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO-L476

**Toolchain:**
GCC_ARM

**Toolchain version:**
To be updated by the customer

**mbed-cli version:** 
To be updated by the customer
(`mbed --version`)

**mbed-os sha:**
To be updated by the customer
(`git log -n1 --oneline`)

**DAPLink version:**
To be updated by the customer

**Expected behavior**
Arm mbed-os should support NUCLEO-L476 UART with 921600 baud rate.

**Actual behavior**
Arm mbed-os does not support NUCLEO-L476 UART with 921600 baud rate.
Using different OS works well with 921600 baud rate and the spec sheet says 1M bps is supported.

**Steps to reproduce**
To be updated by the customer

jeromecoutant, 2017-12-19:
Hi
Could you try #5570 to check the new behavior ?
Thx

andrewc-arm, 2017-12-19:
@lelely2, could you elaborate and update the sections with "To be updated by the customer"?

cmonr, 2017-12-19:
@andrewc-arm @lelely2 We're going to need more information for the issue. I confirmed that serial @ 921600 baud is working fine with a modified blinky example (`mbed import`) as well as the latest mbed release (`mbed update`).

cmonr, 2017-12-19:
@andrewc-arm I'll be expecting more information on how to reproduce the issue within a day or two since this is a blocking issue. Otherwise, feel free to update the priority as need.

lelely2, 2017-12-20:
I use MDK-ARM.
How do I update mbed?

andrewc-arm, 2017-12-20:
Hi. I just talked with @lelely2 about how to test the PR candidate source code (https://github.com/TeamPorcupine/ProjectPorcupine/wiki/How-to-Test-a-Pull-Request)

lelely2, 2017-12-21:
Hi,
Could you give me working sample code for MDK-ARM?
Let me test it.

andrewc-arm, 2017-12-22:
Hi. @lelely2. Could you please try the whole repo of https://github.com/jeromecoutant/mbed/tree/PR_STDIO?

cmonr, 2018-07-26:
Closing since could not reproduce and no updated were given.

Please open another issue if this is still a problem

closed 2018-07-26 by cmonr