ARMmbed/mbed-os #5735: Silicon Labs RAILv2 update missing efr32xg12 configurations

ryankurte, 2017-12-19:
PR #5579 appears to be missing configurations for efr32xg12 (and efr32xg13?) Sub1GHz operation.

@stevew817

stevew817, 2017-12-20:
Thanks for the poke, @ryankurte. I'll look into this.
EFR32xG13 hasn't been ported to mbed by us. We don't have any low cost development kit for it either. Is there any specific reason you're mentioning it here?

ryankurte, 2017-12-20:
Thanks! I haven't got hands on any yet, either. Sounds like they might be
an option for the project I'm working on though.
Only mention it because might as well add the config for them if they're
stable, but not if not.

ryankurte, 2018-01-07:
@stevew817 having some build issues on one platform (Linux) due to duplicated libraries / configs, doesn't appear on OSX for unknown reasons. Should the `TOOLCHAIN_GCC_ARM` folder be removed now RAIL is toolchain independent?

```
➜  [11:22] mbed-router git:[master] ls mbed-os/targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/TARGET_EFR32_1*
mbed-os/targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/TARGET_EFR32_1:
ieee802154_subg_efr32xg1_configurator_out.c ieee802154_subg_efr32xg1_configurator_out.h librail_efr32xg1_release.a

mbed-os/targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/TARGET_EFR32_12:
ieee802154_subg_efr32xg12_configurator_out.c ieee802154_subg_efr32xg12_configurator_out.h librail_efr32xg12_release.a
```

```
➜  [11:23] mbed-router git:[master] ls mbed-os/targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/TOOLCHAIN_GCC_ARM/*
mbed-os/targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/TOOLCHAIN_GCC_ARM/TARGET_EFR32FG1:
ieee802154_efr32xg1_configurator_out.h      ieee802154_subg_efr32xg1_configurator_out.h
ieee802154_subg_efr32xg1_configurator_out.c librail_efr32xg1_gcc_release.a

mbed-os/targets/TARGET_Silicon_Labs/TARGET_SL_RAIL/efr32-rf-driver/rail/TOOLCHAIN_GCC_ARM/TARGET_EFR32FG12:
ieee802154_efr32xg12_configurator_out.h librail_efr32xg12_gcc_release.a
```

stevew817, 2018-02-05:
The decision was made internally to not prioritize validation of Sub-GHz radio configurations. Therefore, I do not have any to commit to the project at this point, and unvalidated configurations, while they might appear to work, might also lead to out-of-regulation operations. This is why I can't commit those either.
I'll submit a PR to remove sub-GHz support from the driver for the time being. I agree it doesn't make sense to have it there but have the xG12 configurations missing. We don't have an official mbed target capable of sub-GHz either, which also played a part in this decision. I hope you understand the reasoning behind this.

ryankurte, 2018-02-06:
That's pretty disappointing tbh, Sub-GHz operation is the reason we've been working on the MBedOS and the Silabs cores. I can almost understand not shipping devices that don't comply, except that they're development tools and compliance with regulations is and has always been on the implementer anyway :-(

It'd be much better for us if you /didn't/ remove the Sub-GHz code. Should we also assume based on this that the same support in RAIL is transient? We made a bet on the next-gen silabs radios some years ago, but, given this and the binary RAIL distribution it's seeming pretty high risk for us to continue with :-/

Sorry if this comes off as a bit grumpy, I'm super appreciative of the work that you've put in and the help that you've given, but, very frustrated to be in this position. I'll have a chat with our account manger and see if anything can be improved ¯\_(ツ)_/¯

stevew817, 2018-02-08:
I understand your frustration. Please do contact your account manager.

No, support in RAIL is not transient. RAIL, and the Flex SDK, will always continue to work with Sub-GHz bands as long as your device contains a Sub-GHz radio. The thing is, and you've probably already noticed it, that the Radio Configurator, which is part of Simplicity Studio, is responsible for the generation of radio configuration blobs. The tool works just fine for proprietary protocols.
In the cases where RAIL accelerates 802.15.4 protocol operation, though, there is some magic happening in those blobs under the hood which requires manual tuning. It is exactly that magic which isn't being prioritised. I can generate bare Radio Configurator output with 802.15.4 PHY parameters, and it might work, but there's also a chance it doesn't. And I don't have the test equipment to properly check the behaviour.

cmonr, 2018-07-26:
Closing since this looks like it won't be resolved.

closed 2018-07-26 by cmonr