ARMmbed/mbed-os #5741: Add support for including/building source/asm files specific to architecture versions

SenRamakri, 2017-12-20:
Enhancement
---------------------------------------------------------------
## Enhancement
Add support for including/building (Cortex-M)Architecture specific source/asm files

**Toolchain:**
GCC_ARM|ARM|IAR

**Suggested enhancement**
Currently there is no way to include a source/asm file specific to a Cortex-M architecture version.
As of now, if we need to do anything specific to an ARCH version we need to replicate the same code across multiple cores under same architecture which creates maintenance issues. For example, we should be able to place code under TARGET_V7M and should get included for all V7M targets. ( Same goes for TARGET_V6M, TARGET_V8M )

**Pros**
Less code duplication

**Cons**

theotherjimmy, 2017-12-20:
The only con I can think of is:
 * Slight increase in build system complexity.

theotherjimmy, 2017-12-20:
@sg- What do you think?

SenRamakri, 2017-12-20:
@theotherjimmy - Yep, I agree, adds little more complexity to build system. On the good side, we can avoid code some replication. For example, Both mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\TOOLCHAIN_ARM\TARGET_M0 and mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\TOOLCHAIN_ARM\TARGET_M0P contains same file. With this support we can move it to TARGET_V6M and should get picked up for both M0 and M0P.

SenRamakri, 2017-12-20:
@theotherjimmy -  I think we should have following folders to support various M archs. I'm also thinking we may need 3 folders for v8 - one for all V8Ms, one for v8m-Baseline only and the other for v8m-Mainline only, because Mainline ISA contains additional instructions not supported by BASELINE. What do you think?

TARGET_V6M
TARGET_V7M
TARGET_V8M
TARGET_V8M_BASELINE
TARGET_V8M_MAINLINE

sg-, 2017-12-20:
> Pros
> Less code duplication

Is there an example of what this would fix?

SenRamakri, 2017-12-20:
@sg- @theotherjimmy  - Yes, For example, both mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\TOOLCHAIN_ARM\TARGET_M0 and mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\TOOLCHAIN_ARM\TARGET_M0P contains same file. With this support we can move it to TARGET_V6M and should get picked up for both M0 and M0P.

sg-, 2017-12-20:
Lets track the blames to see where/why those were introduced.

theotherjimmy, 2017-12-21:
@sg- @SenRamakri 
```
# git log rtos/TARGET_CORTEX/rtx5/RTX/Source/TOOLCHAIN_ARM/TARGET_M0

commit b8aa068def5ae622b5a500cce0e19b771c501025
Author: Bartek Szatkowski <bulislaw@linux.com>
Date:   Wed Oct 25 11:13:51 2017 -0500

    CMSIS/RTX: Rename asm files to upper case .S

commit a03591d6e3b4a441199c73a342dcf70156b9ae07
Author: Bartek Szatkowski <bulislaw@linux.com>
Date:   Thu Oct 12 13:05:25 2017 -0500

    CMSIS/RTX: Update CMSIS and RTX to 22b68c
    
    This includes Cortex A support and directory reshuffle.
# git log rtos/TARGET_CORTEX/rtx5/RTX/Source/TOOLCHAIN_ARM/TARGET_M0P
commit b8aa068def5ae622b5a500cce0e19b771c501025
Author: Bartek Szatkowski <bulislaw@linux.com>
Date:   Wed Oct 25 11:13:51 2017 -0500

    CMSIS/RTX: Rename asm files to upper case .S

commit a03591d6e3b4a441199c73a342dcf70156b9ae07
Author: Bartek Szatkowski <bulislaw@linux.com>
Date:   Thu Oct 12 13:05:25 2017 -0500

    CMSIS/RTX: Update CMSIS and RTX to 22b68c
    
    This includes Cortex A support and directory reshuffle.

SenRamakri, 2017-12-21:
@sg- @theotherjimmy  - Tried digging little more into git history and it appears this folder structure/replication has been in existence for quite a while, although it has been moved around few times as shown in Jimmy's git log above. For example, this commit( SHA ID: d9734e5a3254ae7ed3ffcb0b7ac513eef7bf0d34) from May-2016 has TARGET_M0/TARGET_M0P replicating the files.

0xc0170, 2017-12-22:
That is correct, we had core labels in the build system. It started M3,then M0, M4 , etc so that lead early to generalizing it as core. Thus we use core labels at the moment.

We would benefit from having arch labels - in our code base it would be in rtos folder. Is there anything else that would be simplified?

closed 2019-10-31 by ciarmcom