ARMmbed/mbed-os #5759: Nanostack memory area (nanostack-heap)

daniel-cesarini-tridonic-com, 2017-12-22:
### Description
- Type: Question
- Priority: Major 

## Question
The nanostack is having a "private" memory section (at least in version 7.0.3), that it is managing and using as a heap. It is being initialized in ```ns_hal_init``` or ```ns_dyn_mem_init```. 
With     ```const mem_stat_t *heap_info = ns_dyn_mem_get_mem_stat(); ``` I can get information about the status of such memory area: ```Heap size```  ```Allocated```  ```Allocated max``` ```#Alloc fail (heap_alloc_fail_cnt)```. 

**I am wondering what the meaning of alloc fail is?**

And furthermore:

**Does it mean that the nanostack is able to handle the case in which that are alloc failures and continue to work, or does it mean that if at any point in time I see at least 1 alloc fail it means that the memory area is too tiny?**

0xc0170, 2017-12-26:
@SeppoTakalo @kjbracey-arm please review

SeppoTakalo, 2018-01-03:
Hi @daniel-cesarini-tridonic-com 

Nanostack is designed to tolerate some level of allocation failures. Internally it buffers all incoming traffic as long as there is memory left. Therefore the memory is set to statically allocated block to give limits to maximum memory usage.
Packet drops are starting to happen once the stack runs out of memory and this is the way of rate limiting the traffic on busy network.

For estimating the optimal size of this heap, these `mem_stat_t` fields are to be used. `heap_alloc_fail_cnt` is a counter that tells how many times the allocation have failed.
On a network with no application layer traffic, this value should be zero. It would tell the developer that on the given memory limit and network settings, devices are able to join the network and complete all the handshake phases.

As these devices are operating on the mesh network and routing traffic originating from other devices we will be seeing increase on memory usage once the network size is rising and devices start to send more application layer traffic.

On a general guideline, allocation failures should stay roughly on zero after network is fully booted up. Small increase on value is OK when all devices initiate their handshake with cloud. These packet drops are handled at the upper level and they rate-limit the joining processes. Once the application traffic has stabilised the value should not continue to increase. If increase is seen, it is a symptom of too much traffic on the mesh side and mean lots of packet drops.

MarceloSalazar, 2018-04-19:
@daniel-cesarini-tridonic-com 

Hope @SeppoTakalo 's response answers your question - let us know if you need more information.

closed 2018-04-19 by MarceloSalazar

kjbracey-arm, 2018-04-30:
The memory needed to "work at all" is largely allocated up front and stays allocated. We did some work after fuzz testing to ensure that was the case - making sure the stack kept running in the event of heap full. Back in the day it might have failed to rescheduled a periodic timer, for example - a non-recoverable situation.

So the active heap contention is a mixture of various transient data like packets and longer-term data like SLAAC address allocations. I guess a separate packet buffer limit would guarantee clearance for routing/address info.

As a quick change I'd probably implement that as a separate audited limit on buffer_alloc() without actually changing the heap structure. Or maybe it could be a limit on `ns_dyn_mem_temporary_alloc()` - make sure it leaves a certain amount free. That sounds like not a bad idea - a new use for the longterm/temporary split we already have.

daniel-cesarini-tridonic-com, 2018-04-30:
@kjbracey-arm thank you for the reply.
I have one question about it:

> As a quick change I'd probably implement that as a separate audited limit on buffer_alloc() without actually changing the heap structure.

how to implement that?

Furthermore,

> Or maybe it could be a limit on ns_dyn_mem_temporary_alloc() - make sure it leaves a certain amount free 

what do you mean by that?

kjbracey-arm, 2018-04-30:
All allocations to the heap ultimately come in to `ns_dyn_mem_alloc` and `ns_dyn_mem_temporary_alloc`. Packet buffers use the latter. The split is primarily intended to avoid fragmentation - they allocate from opposite ends, meaning long-term heap blocks don't cause holes left due to allocating on top of a temporary buffers

You could add some logic to `ns_dyn_mem_temporary_alloc` to make sure it doesn't exhaust the heap - just return NULL if the allocation would leave less than some configurable amount free. Which effectively means network packets can't on their own exhaust the heap. The `ns_dyn_mem_alloc` allocations get priority.

daniel-cesarini-tridonic-com, 2018-05-09:
Opened new PR "Split nanostack heap into using 2 different books" : https://github.com/ARMmbed/nanostack-libservice/pull/67