ARMmbed/mbed-os #5762: EFM32: EM2 mode blocked when should not be

Luka234, 2017-12-28:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
EFM32GG_STK3700 

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
8b54959

**Expected behavior**
Calling sleep() should enter lowest energy mode possible. If all used instances of timer classes (Ticker, Timeout, Timer) are currently inactive then they should not prevent entering lower EM modes.

**Actual behavior**
If there is instance of any timer class (Ticker, Timeout, Timer) or using  wait function, sleep() is limited to EM1 (it is blocked by block counter inside hal_sleep function).

**Steps to reproduce**

Example 1: just instantiating Timer object (timer is not started) will block any EM above EM1. Same behavior is with Ticker (without attaching to it) or Timeout.
```
#include "mbed.h"
Timer timer;

int main( void ) {
    while(1) {
        sleep();
    }
}
```

Example 2: First sleep will not be forced to EM1 while any other after calling wait_ms will be forced to EM1.

```
#include "mbed.h"

int main( void ) {
    while(1) {
        sleep();
        wait_ms(100);
    }
}
```

----------------------------------------------------------------
## Enhancement

I'm aware that for Example 1 we could use LowPower variants of timer classes, but that solution is not always possible/practical. 

For Example 2 there is no alternatives that I'm aware of. 

-----------------------------------------------------------------

Luka234, 2017-12-29:
Sorry guys, false alarm. Deep sleep functionality from sleep manager is handling scenarios that I have described in a bug report.

closed 2017-12-29 by Luka234

Luka234, 2017-12-29:
Is someone encounter similar (not existing, or maybe not as severe as I initially though) problem. 
In debug mode system will behave as I've described, but in release it is ok.
EFM32 has sleep block functionality inside sleep.c that is working even in debug mode (MBED_DEBUG defined) so I though that it will be same sleep behavior as in release. 
Mbed itself has another deep sleep block functionality in mbed_sleep_manager.c which is disabled during debug mode. In release mode that function is enabled and EFM32 will enter EM2 (fixed, no lower modes) if it can. Monitoring if timers are active is done in Mbed deep sleep and not in EFM32 sleep.c and that is the reason why I have observed locking of energy modes lower than EM1.
For current EFM32 sleep functionality it would be safe that mbed_sleep_manager.c is calling hal_deepsleep in debug mode - but that is improvement and not a bug.

0xc0170, 2018-01-08:
> For current EFM32 sleep functionality it would be safe that mbed_sleep_manager.c is calling hal_deepsleep in debug mode - but that is improvement and not a bug.

That is the intention, as some targets might not have debugger available in deep sleep or some other parts not clocked, therefore in debug only sleep is available.