ARMmbed/mbed-os #5763: nrf52-dk: Unable to export to gnuarmeclipse

aldrinleal, 2017-12-28:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

**Target**
nrf52_dk

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)
1.3.0

**mbed-os sha:**
(`git log -n1 --oneline`)
any

**DAPLink version:**

**Expected behavior**
Should Compile on Eclipse

**Actual behavior**

Dumps these error messages:

```
Building file: ../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/lp_ticker.c
Invoking: Cross ARM C Compiler
arm-none-eabi-gcc -mcpu=cortex-m4 -mthumb -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -O0 -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin -funsigned-char -fno-delete-null-pointer-checks -fomit-frame-pointer -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers  -g3 -DDEVICE_ERROR_PATTERN=1 -DTARGET_MCU_NRF52832 -DTARGET_SDK11 -D__MBED__=1 -DNRF52_PAN_53 -DTARGET_NRF52_COMMON -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DBOARD_PCA10040 -DTARGET_FF_ARDUINO -DNRF52_PAN_12 -DTOOLCHAIN_GCC -DNRF52_PAN_58 -DDEVICE_SPISLAVE=1 -DNRF52_PAN_55 -DNRF52_PAN_54 -DTARGET_CORTEX_M -DTARGET_DEBUG -DTARGET_NRF52_DK -DNRF52_PAN_36 -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_MCU_NRF52 -DCMSIS_VECTAB_VIRTUAL -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DNRF52 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DTARGET_NRF52832 -D__CORTEX_M4 -D__FPU_PRESENT=1 -DS132 -DDEVICE_SLEEP=1 -DDEVICE_PORTIN=1 -DNRF52_PAN_51 -DTARGET_NORDIC -DNRF52_PAN_30 -DBLE_STACK_SUPPORT_REQD -DNRF52_PAN_15 -DCONFIG_GPIO_AS_PINRESET -DMBED_BUILD_TIMESTAMP=1514470223.58 -DTARGET_LIKE_MBED -D__MBED_CMSIS_RTOS_CM -DNRF52_PAN_31 -DTOOLCHAIN_GCC_ARM -DSWI_DISABLE0 -DSOFTDEVICE_PRESENT -DDEVICE_I2C_ASYNCH=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\" -DDEVICE_INTERRUPTIN=1 -DNRF52_PAN_64 -DNRF52_PAN_20 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DNRF52_PAN_62 -DNRF52_PAN_63 -DDEVICE_SPI=1 -DARM_MATH_CM4 -DDEVICE_SERIAL_FC=1 -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/." -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/DHT11" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/DHT11/mbed" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/DHT11/mbed/e188a91d3eaa" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/rtos" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/rtos/rtx4" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/rtos/rtx5" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/rtos/rtx5/TARGET_CORTEX_M" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/rtos/rtx5/TARGET_CORTEX_M/TARGET_RTOS_M4_M7" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/rtos/rtx5/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_GCC" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/drivers" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/nanostack" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/storage" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/utest" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/utest/utest" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/utest/source" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/unity" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/unity/unity" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/unity/source" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/greentea-client" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/greentea-client/greentea-client" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/frameworks/greentea-client/source" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/filesystem" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/filesystem/bd" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/filesystem/fat" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/filesystem/fat/ChaN" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/netsocket" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/netsocket/cellular" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/netsocket/cellular/utils" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/netsocket/cellular/generic_modem_driver" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls/importer" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls/platform" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls/platform/inc" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls/platform/src" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls/targets" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls/inc" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls/inc/mbedtls" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/features/mbedtls/src" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/platform" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/hal" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/hal/storage_abstraction" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_NRF52_COMMON" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk_patch" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/TARGET_NRF52_DK" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/driver_nrf" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/driver_nrf/saadc" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/driver_nrf/pwm" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/softdevice" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/softdevice/s132" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/softdevice/s132/headers" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/softdevice/s132/headers/nrf52" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht11/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/sdk/softdevice/s132/hex" -I"/home/aldrin/projetos/sources/mbed-work/test-nrf52-dht [... cut ...]
make: *** [mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/i2c_api.o] Error 1
Finished building: ../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/gpio_api.c
 
make: *** Waiting for unfinished jobs....
Finished building: ../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/flash_api.c
 
Finished building: ../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/lp_ticker.c
 

09:11:13 Build Finished (took 10s.857ms)
```

**Steps to reproduce**

 * Create an new mbed project
 * mbed target nrf52_dk
 * mbed toolchain GCC_ARM
 * mbed export -i gnuarmeclipse
 * Import on eclipse. Build.

0xc0170, 2018-01-02:
Just in case, building with GCC_ARM using `mbed compile` works

Based on the errors seems like it does not use correct i2c structures (might be missing some configuration like i2c async or similar) ?

aldrinleal, 2018-01-02:
@0xc0170 yes, mbed compile works. I wonder whats wrong with it

theotherjimmy, 2018-04-23:
@aldrinleal We have been testing this particular export combination, and a recent run pased: http://mbed-os.s3-website-eu-west-1.amazonaws.com/?prefix=builds/exporter/6447/PASS/NRF52_DK/

theotherjimmy, 2018-04-23:
Full log here: http://mbed-os.s3-eu-west-1.amazonaws.com/builds/exporter/6447/PASS/NRF52_DK/gnuarmeclipse/f9ed86849fd49103c75982e3bc00834760b61666_exporter_build_log_NRF52_DK_gnuarmeclipse.txt

closed 2018-04-23 by theotherjimmy

aldrinleal, 2018-04-23:
Awesome! Thank you guys!