ARMmbed/mbed-os #5765: STM can_filter not filter right for using wrong mask bit

zh0umu, 2017-12-29:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32

**Description**
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/can_api.c
mask bit 0x00FF as code below  is wrong. Id and mask low part is 16bit.So it should be 0xFFFF
sFilterConfig.FilterIdHigh = id >> 13; // EXTID[28:13]
        sFilterConfig.FilterIdLow = (0x00FF & (id << 3)) | (1 << 2);  // EXTID[12:0]
            sFilterConfig.FilterMaskIdHigh = mask >> 13;
            sFilterConfig.FilterMaskIdLow = (0x00FF & (mask << 3)) | (1 << 2);
**Expected behavior**
filter should only receive message accroding to id and mask
**Actual behavior**
filter not working with cetain filter

bcostm, 2018-01-02:
Thanks for this analysis.

Can you send a PR with your correction ? And can you post also here a test program that we can use to test it ?

Related issue: #5351

bcostm, 2018-01-04:
Looking in the can_api.c file, I would change this:

 ```
    if (format == CANStandard) {
            sFilterConfig.FilterIdHigh = id << 5;
            sFilterConfig.FilterIdLow =  0x0;
            sFilterConfig.FilterMaskIdHigh = mask << 5;
            sFilterConfig.FilterMaskIdLow = 0x0; // allows both remote and data frames
        } else if (format == CANExtended) {
            sFilterConfig.FilterIdHigh = id >> 13; // EXTID[28:13]
            sFilterConfig.FilterIdLow = (0x00FF & (id << 3)) | (1 << 2);  // EXTID[12:0]
            sFilterConfig.FilterMaskIdHigh = mask >> 13;
            sFilterConfig.FilterMaskIdLow = (0x00FF & (mask << 3)) | (1 << 2);
        }
```

with this:

```
            sFilterConfig.FilterIdHigh = id >> 16;
            sFilterConfig.FilterIdLow =  id & 0xFFFF;
            sFilterConfig.FilterMaskIdHigh = mask >> 16;
            sFilterConfig.FilterMaskIdLow = mask & 0xFFFF;
```

Can you please check this code and tell me if it solves your issue ? Thanks.

zh0umu, 2018-01-05:
@bcostm I change with below，working ok.
            sFilterConfig.FilterIdHigh = id >> 13; // EXTID[28:13]
            sFilterConfig.FilterIdLow = (0xFFFF & (id << 3)) | (1 << 2);  // EXTID[12:0]
            sFilterConfig.FilterMaskIdHigh = mask >> 13;
            sFilterConfig.FilterMaskIdLow = (0xFFFF & (mask << 3)) | (1 << 2);

bcostm, 2018-01-05:
I thought the ST HAL HAL_CAN_ConfigFilter function managed everything concerning the ID and Mask bits... But it is not the case. So, your fix looks OK to me. I'll send a PR then.

jeromecoutant, 2018-01-24:
As #5791 is now merged, I think we can close the issue ?
Thx

closed 2018-01-24 by 0xc0170