ARMmbed/mbed-os #5769: Cannot export microbit programs in online compiler

janjongboom, 2017-12-30:
@theotherjimmy 

### Description
- Type: Bug

## Bug

**Target**
BBC micro:bit

**Toolchain:**
ARM

**Expected behavior**

Exporting works.

**Actual behavior**

Exporting does not work

**Steps to reproduce**

See https://os.mbed.com/questions/79775/BBC-microbit-DAL-example-project-wont-ex/

carlosperate, 2017-12-30:
Thank you for reporting this @janjongboom!
Just to give a bit more extra info, the export does complete for the VSCode + ARMC compiler, but not for VSCode + ARM GCC.

0xc0170, 2018-01-02:
> Just to give a bit more extra info, the export does complete for the VSCode + ARMC compiler, but not for VSCode + ARM GCC.

Is this case also for local export, if you fetch blinky example and do it locally, will it export? to better understand if this is a problem in our scripts (online IDE does not shows log, you can see them if you use mbed-cli). Please report

theotherjimmy, 2018-01-02:
@janjongboom @carlosperate Could you post a link to the example you are describing?

theotherjimmy, 2018-01-02:
FYI: the microbit does not support IAR, so don't expect to be able to export to an IAR-using IDE.

carlosperate, 2018-01-05:
I was just trying the default micro:bit hello workd: https://os.mbed.com/teams/microbit/code/microbit-hello-world/

My published version: https://os.mbed.com/users/carlosperate/code/microbit-hello-world/

On Mbed CLI vs 1.3.0:

```
> mbed import https://os.mbed.com/users/carlosperate/code/microbit-hello-world/
... 
> mbed compile -t GCC_ARM -m NRF51_MICROBIT
... 
(Have issues compiling here [compiles fine online], but it detects the compiler correctly, so putting this aside at the moment)
...
> mbed export -i vscode_gcc_arm -m NRF51_MICROBIT
Scan: .
[ERROR] No linker script found.
```

Does that help?

theotherjimmy, 2018-01-08:
@carlosperate Yep! That helps for sure. Your testing indicates that GCC_ARM is not supported by the [build of the mbed-dev-bin](https://os.mbed.com/teams/Lancaster-University/code/mbed-dev-bin/file/768173a57492/TARGET_NRF51_MICROBIT/) that is used by that project.

theotherjimmy, 2018-01-08:
@thegecko We should discuss a way to pass this error along to the online compiler front end, if possible.

theotherjimmy, 2018-01-08:
@carlosperate You may be having many "symbol <x> not found" errors, as the build of mbed-dev-bin does not contain a static library for your compiler.

carlosperate, 2018-01-14:
Thanks @theotherjimmy , that explains the linking issue I was having. So is that the reason the export fails?

theotherjimmy, 2018-01-22:
@carlosperate Yes, that's related to reason the export fails. The export fails because it can't find a linker script for your target, which would live next to the static library containing the pile-o-missing-symbols.

theotherjimmy, 2018-09-24:
Closing as these errors are now reported by the online compiler.

closed 2018-09-24 by theotherjimmy

jxgx072037, 2018-10-09:
The mbed team is working on that issue, https://os.mbed.com/questions/82626/Compiling-for-MicroBit-No-Linker-Script-/