ARMmbed/mbed-os #5771: Add support for Eclipse Che

amq, 2018-01-02:
### Description
- Type: Enhancement
- Priority: Minor

## Enhancement

[Eclipse Che](https://www.eclipse.org/che/) is an open source online IDE which could be a great addition to the Mbed OS development ecosystem. It is also already being used for [embedded development](http://www.eclipse.org/che/docs/tutorials/platformio/index.html).

loverdeg-ep, 2018-01-04:
I'm VERY EAGERLY waiting for this as well. 
However I think it's going to block until there is further development of either [clangd](https://clang.llvm.org/extra/clangd.html) or [cquery](https://github.com/jacobdufault/cquery) language servers. 

Support for remotely loading and debugging a target in Che will be needed as well.
I beleive platformIO does remote debug but is a paid feature and not enable for Che last I checked.

The debugging and test tools/servers the mbed team are developing should work rather nicely with a
[workspace server like Che](https://www.eclipse.org/che/docs/#logical-architecture).

Other Relevant Links:
https://www.eclipse.org/community/eclipse_newsletter/2017/april/article5.php
https://www.eclipse.org/community/eclipse_newsletter/2017/may/article1.php

mark-edgeworth, 2019-06-12:
I'm sorry but we don't have the bandwidth here to consider implementing this. You might consider our Mbed Studio product which is based on Eclipse Theia and may well offer the features that you are looking for.

closed 2019-06-12 by mark-edgeworth

loverdeg-ep, 2019-06-12:
Final note. 

* Mbed Studio is based on Theia. 
* Theia is Che's default IDE.
* Che also apparently supports VS Code plugins out of the box.

I have no say in this but depending on where the Theia project goes, Che may or may not become an emergent path as Mbed Studio evolves. Either way I can only hope the end result is the same.

@JoeTheGuitarist