ARMmbed/mbed-os #5772: Wrong default stats-depth parameter on Windows

MarceloSalazar, 2018-01-02:
### Description
The memory map output generated by memap has different length, depending on the host OS you use. I suspect the problem is caused by the default `--stats-depth` parameter not being passed correctly on Windows. 
- Type: Bug
- Priority: Major / Minor

## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
(`mbed --version`)
1.3.0

**mbed-os sha:**
(`git log -n1 --oneline`)
2b4ff78ab0a52ef1dc3f2998908453c595e2b2c0

**Expected behavior**

Length of memory map output should be the same, regardless of using Windows or OSX host OS.

**Actual behavior**

Windows:

```
+--------------------------------------------------------------------------------------------------------------------+-------+-------+------+
| Module                                                                                                             | .text | .data | .bss |
+--------------------------------------------------------------------------------------------------------------------+-------+-------+------+
| [fill]                                                                                                             |    78 |     4 | 2574 |
| [lib]/c.a                                                                                                          | 22929 |  2472 |   89 |
| [lib]/gcc.a                                                                                                        |  3144 |     0 |    0 |
| [lib]/misc                                                                                                         |   248 |     8 |   28 |
| [lib]/nosys.a                                                                                                      |    32 |     0 |    0 |
| main.o                                                                                                             |    56 |     0 |    4 |
| mbed-os\features\filesystem\littlefs\TESTS_COMMON\atomic_usage.o                                                   |    65 |     0 |    4 |
| mbed-os\features\storage\FEATURE_STORAGE\cfstore\source\cfstore_svm.o                                              |    42 |     0 |  184 |
| mbed-os\hal\mbed_gpio.o                                                                                            |    96 |     0 |    0 |
| mbed-os\hal\mbed_pinmap_common.o                                                                                   |   212 |     0 |    0 |
| mbed-os\hal\mbed_sleep_manager.o                                                                                   |   219 |     0 |    2 |
| mbed-os\hal\mbed_ticker_api.o                                                                                      |   840 |     0 |    0 |
| mbed-os\hal\mbed_us_ticker_api.o                                                                                   |    64 |     4 |   64 |
| mbed-os\platform\mbed_alloc_wrappers.o                                                                             |    16 |     0 |    0 |
| mbed-os\platform\mbed_assert.o                                                                                     |    85 |     0 |    0 |
| mbed-os\platform\mbed_board.o                                                                                      |   204 |     0 |    0 |
| mbed-os\platform\mbed_critical.o                                                                                   |   305 |     0 |    5 |
| mbed-os\platform\mbed_error.o                                                                                      |    44 |     0 |    1 |
| mbed-os\platform\mbed_retarget.o                                                                                   |   658 |     4 |  264 |
| mbed-os\platform\mbed_wait_api_rtos.o                                                                              |   102 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\mbed_boot.o                                                                             |   459 |     0 | 4360 |
| mbed-os\rtos\TARGET_CORTEX\mbed_rtx_handlers.o                                                                     |   663 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\mbed_rtx_idle.o                                                                         |    42 |     4 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\TOOLCHAIN_GCC\TARGET_RTOS_M4_M7\irq_cm4f.o                              |   196 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rt_OsEventObserver.o                                                    |     0 |     0 |    4 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_delay.o                                                             |    72 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_evr.o                                                               |   108 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_kernel.o                                                            |   871 |   164 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_lib.o                                                               |   260 |     0 | 1620 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_memory.o                                                            |   242 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_mempool.o                                                           |   288 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_msgqueue.o                                                          |  1708 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_mutex.o                                                             |   832 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_system.o                                                            |   378 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_thread.o                                                            |  2066 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\RTX\Source\rtx_timer.o                                                             |   216 |     0 |    0 |
| mbed-os\rtos\TARGET_CORTEX\rtx5\Source\os_systick.o                                                                |   126 |     0 |    1 |
[... 34 lines cut ...]
| [fill]           |    74 |     4 | 2574 |
| [lib]/c.a        | 22917 |  2472 |   89 |
| [lib]/gcc.a      |  3144 |     0 |    0 |
| [lib]/misc       |   248 |     8 |   28 |
| [lib]/nosys.a    |    32 |     0 |    0 |
| main.o           |    56 |     0 |    4 |
| mbed-os/features |   107 |     0 |  188 |
| mbed-os/hal      |  1431 |     4 |   66 |
| mbed-os/platform |  1414 |     4 |  270 |
| mbed-os/rtos     |  8547 |   168 | 5989 |
| mbed-os/targets  |  8119 |    12 |  384 |
| Subtotals        | 46089 |  2672 | 9592 |
+------------------+-------+-------+------+
Total Static RAM memory (data + bss): 12264 bytes
Total Flash memory (text + data): 48761 bytes

```

**Steps to reproduce**

```
mbed import mbed-os-example-blinky
cd mbed-os-example-blinky
mbed compile -t GCC_ARM -m K64F
```

theotherjimmy, 2018-01-02:
@MarceloSalazar Thanks for the bug report. This is a "we did not use the path seperator" bug. I'll put up a patch soonish (next day or two) to fix this.

theotherjimmy, 2018-01-02:
I'd like to point out: it's not that the stats depth is set incorrectly on windows as the libraries seem to heed the depth setting. It's that the depth was splitting on `/` instead of `os.sep`, and that created problems for non-library modules when the `os.sep is not /`, basically on only windows.

PR #5775 (2018-01-08): Correct memap path handling on windows
Includes a parser refactor, as I got a bit irritated by seeing `_gcc`
suffixed method names.

Resolves #5772

# Todo
 - [x] A test for path handling
 - [x] A test for GCC including builtin parsing
 - [x] A test for multi-parameter cmd line parsing with IAR
closed 2018-01-08 by theotherjimmy